PX4/PX4-Autopilot #4493: operator new: exceptions vs nullptr

bkueng, 2016-05-10:
I noticed strange behavior with the following simple code:
```
bool LogWriter::init()
{
	if (_buffer)
		return true;

	_buffer = new uint8_t[_buffer_size];
	return _buffer;
}
```
Even if the allocation failed, the function returned true. So I looked at the disassembly:
```
00000000 <px4::logger::LogWriter::init()>:
   0:	b510      	push	{r4, lr}
   2:	6c43      	ldr	r3, [r0, #68]	; 0x44
   4:	4604      	mov	r4, r0
   6:	b91b      	cbnz	r3, 10 <px4::logger::LogWriter::init()+0x10>
   8:	6c80      	ldr	r0, [r0, #72]	; 0x48
   a:	f7ff fffe 	bl	0 <operator new[](unsigned int)>
   e:	6460      	str	r0, [r4, #68]	; 0x44
  10:	2001      	movs	r0, #1
  12:	bd10      	pop	{r4, pc}
```
And indeed the function returns unconditionally 1 (line 10). I guessed the compiler assumes that operator new does not return nullptr (as it would throw std::bad_alloc). I then compiled with `-fcheck-new`, which disables the assumption, and the result is:
```
00000000 <px4::logger::LogWriter::init()>:
   0:	b510      	push	{r4, lr}
   2:	6c43      	ldr	r3, [r0, #68]	; 0x44
   4:	4604      	mov	r4, r0
   6:	b93b      	cbnz	r3, 18 <px4::logger::LogWriter::init()+0x18>
   8:	6c80      	ldr	r0, [r0, #72]	; 0x48
   a:	f7ff fffe 	bl	0 <operator new[](unsigned int)>
   e:	6460      	str	r0, [r4, #68]	; 0x44
  10:	3000      	adds	r0, #0
  12:	bf18      	it	ne
  14:	2001      	movne	r0, #1
  16:	bd10      	pop	{r4, pc}
  18:	2001      	movs	r0, #1
  1a:	bd10      	pop	{r4, pc}
```
And this is correct. I think this affects the whole code base, so we need to add `-fcheck-new` to the cxx flags (right next to `-fno-exceptions`). Or declare the operator new with `throw()`.

LorenzMeier, 2016-05-10:
Thanks! Can you send a PR with that?

LorenzMeier, 2016-05-10:
Fixed.

closed 2016-05-10 by LorenzMeier