PX4/PX4-Autopilot #4498: Multicopter position controller integrates error when sitting idle without being airborne

LorenzMeier, 2016-05-10:
@tumbili The issue seems that when armed and in hold flight mode a position setpoint integral is building up, which then first needs to be unwound when taking off in takeoff mode.

http://logs.uaventure.com/view/pvEC9wamaU8QcmDwExzFw5

Particularly here:
http://logs.uaventure.com/view/pvEC9wamaU8QcmDwExzFw5#Local_X_PLOT

LorenzMeier, 2016-05-12:
@tumbili Can you please set your priority on this one? I think we need to fix this for 1.3.0 immediately.

tumbili, 2016-05-12:
The problem here is that the navigator publishes zero values for lat/lon in the position setpoint.
Another potential problem (not the main issue) is that we don't reset position setpoints in non-manual modes when we are landed. This is addressed here https://github.com/PX4/Firmware/pull/4514

LorenzMeier, 2016-05-12:
Do we know if it publishes a zero SP?

tumbili, 2016-05-12:
@LorenzMeier At least my printf tells me that. Lat/lon are both zero and I verified that the topic updates.

LorenzMeier, 2016-05-12:
ok, thanks!

LorenzMeier, 2016-05-13:
Fixed in a700b02.

closed 2016-05-13 by LorenzMeier