PX4/PX4-Autopilot #4500: Feature Request Pitch angle

jcooper21, 2016-05-10:
SenseFly has a really nice mapping feature that allows you to set your pitch.  Please see below for details on how it works.  It would be awesome if PX4 was able to implement something similar.
[image]

LorenzMeier, 2016-05-10:
Thanks for the request. Its super helpful you're keeping feeding us good user story feedback.

jcooper21, 2016-05-10:
I forgot to note this a fixed wing feature.  I am sure its obvious but making sure there is no mis understanding.

jgoppert, 2016-05-15:
It would also be useful for quads. I have a pan/tilt setup on my 250 pixracer and having it auto level the camera would probably make flying simpler.

TSC21, 2018-02-21:
Still relevant and worth discussion for future implementation.

closed 2019-02-13 by stale[bot]