PX4/PX4-Autopilot #4511: FW pos control bad attitude tracking

tumbili, 2016-05-12:
It seems that for fixed wings the attitude controller does not control the roll axis well in position control mode. In the image below position control mode is active when the red line is at 2.
[image]

LorenzMeier, 2016-05-12:
@tumbili You're confused by the logging. When the user moves the roll stick his actual input is not logged. Only when he re-centers the controls.

tumbili, 2016-05-12:
@LorenzMeier Yeah I just realized that as well

dagar, 2016-05-13:
We should flesh out the details of the stick mapper (https://github.com/PX4/Firmware/issues/2822) and I'll start working on it.

tumbili, 2016-08-15:
Fixed in master

closed 2016-08-15 by tumbili