PX4/PX4-Autopilot #4515: Ekf2: Potential issue with replay logging

CarlOlsson, 2016-05-12:
I have the issue that I can't get non zero airspeed innovations when I use recorded replay data.
- Flashed current master 54b3995175d86bcbb23e8c7e1b94aca9d7e76b35 on the pixracer.
- Did handheld test with replay logging activated ([log1](http://logs.uaventure.com/logmuncher/charts/kVbGQqZZqMLhE5xoD4yni/logmuncher.log))
- Activated airspeed fusion in ecl by uncommenting line 352 in ekf.cpp
- Replayed the data but all airspeed innovations are zero

- Did a new hand held test, now with airspeed fusion activated during the test
- Replayed the log ([log2](http://logs.uaventure.com/logmuncher/charts/YZYZpmy9KZ4U9H9CEFD9n5/logmuncher.log
)) but still all airspeed innovations are zero

- Deactivated replay logging
- Did a new handheld test ([log3](http://logs.uaventure.com/logmuncher/charts/F7CfmyvBnVQJcx3dXnMpTk/logmuncher.log))
- Now I see airspeed innovations

Am I doing something wrong or do you think there a bug in the replay logging?
@tumbili @priseborough 

log1
Plots: http://logs.uaventure.com/view/kVbGQqZZqMLhE5xoD4yni
Log: http://logs.uaventure.com/logmuncher/charts/kVbGQqZZqMLhE5xoD4yni/logmuncher.log

log2
Plots: http://logs.uaventure.com/view/YZYZpmy9KZ4U9H9CEFD9n5
Log: http://logs.uaventure.com/logmuncher/charts/YZYZpmy9KZ4U9H9CEFD9n5/logmuncher.log

log3
Plots: http://logs.uaventure.com/view/F7CfmyvBnVQJcx3dXnMpTk
Log: http://logs.uaventure.com/logmuncher/charts/F7CfmyvBnVQJcx3dXnMpTk/logmuncher.log

CarlOlsson, 2016-05-12:
Ok I looked into the ekf2_replay_main.cpp file and the airspeed is missing. Will add it and create a pr

tumbili, 2016-05-12:
@CarlOlsson I can help in the afternoon

CarlOlsson, 2016-05-12:
PR here: #4518

closed 2016-05-12 by CarlOlsson