PX4/PX4-Autopilot #4516: In the SITL gazebo simu, the MAV runs away.

Userskii, 2016-05-12:
I updated the firmware on Monday then on Wednesday. However after the latest update. After running $ make posix_sitl_default gazebo, the iris does not rise up vertically. It runs away. Did I do something wrong?

julianoes, 2016-05-13:
Thanks for raising. I can reproduce this as well when doing `commander takeoff`. It seems to be broken.

julianoes, 2016-05-13:
@Userskii In the meantime, you should be able to use `make posix jmavsim`.

julianoes, 2016-05-13:
I'm looking into it.

julianoes, 2016-05-13:
@LorenzMeier: `git bisect` points to 8960ab3402a87b182d17ab1b3cb165115e0f4bad

julianoes, 2016-05-13:
Is this connected to https://github.com/PX4/Firmware/pull/4514?

julianoes, 2016-05-13:
This would fix it for now #4533.

closed 2016-05-13 by LorenzMeier

Userskii, 2016-05-16:
I tested and it works. Thanks.

commit 7d664355 (2016-05-13): Commander: Fix the takeoff and land commands. Fixes #4516