PX4/PX4-Autopilot #4523: FW posctrl wing rocking

tumbili, 2016-05-13:
According to http://discuss.px4.io/t/position-control-gain-parameter/413
Log: http://logs.uaventure.com/view/dfztG6auv7PTQsot7ZXGGD

The rollrate setpoint in position control mode is not continuous and seems to be jumping between positive and negative. The roll setpoint (if it's logged correctly) does not do this so we need to investigate the root cause of this.
[image]

LorenzMeier, 2016-05-22:
@tumbili Any news on the investigation of this?

jcooper21, 2016-05-26:
I just did another test flight and assuming you did not push out a patch for this I believe I may have found something.  I tested with a Zubax GNSS 2 just now rather than the Hobby King M8N I was previously using and the issue seems to be almost gone.  Is it a GPS position error causing the issue.  See the zubax log below.

http://logs.uaventure.com/view/9N3JL4kgFvoVUycRRarZbZ#GPS_Noise_PLOT

tumbili, 2016-06-07:
@jcooper21 I haven't changed anything on master and I haven't been able to reproduce this.
Could you do another test flight on master and see if you still have this problem.

jcooper21, 2016-06-08:
@tumbili I will switch out my GPS to the HK M8N and do another flight test. I'll try in the next couple days and upload a log.

tumbili, 2016-06-10:
@jcooper21 Any news on this?

jcooper21, 2016-06-10:
Not yet I have had high winds and rain so I am still waiting for the weather to clear.

jcooper21, 2016-06-14:
I tested this today with the M8N gps and the current beta firmware the results were the rocking issue still exits.  Its extremely bad in winds.  See below log.

http://logs.uaventure.com/view/25mLx5BQ8JwUwidXUCUD7a

tumbili, 2016-06-27:
@jcooper21 Finally found the problem https://github.com/PX4/Firmware/pull/4916

closed 2016-06-27 by tumbili

AndreasAntener, 2016-07-25:
@jcooper21 this is on master now