PX4/PX4-Autopilot #4532: Testing

dagar, 2016-05-13:
@bkueng @LorenzMeier @AndreasAntener let's discuss where we'd like to get with testing and coordinate our effort.

Overall it would be nice to have unit and integration testing that's run on every platform.

Ideas/Todo
 - unit tests (src/systemcmds/tests and unittests)
  - move to gtest and merge them (WIP https://github.com/PX4/Firmware/compare/master...dagar:gtest?expand=1)
  - runs with px4fmu-v2_test (hardware test harness) and posix_sitl_test (travis-ci)
  - modify mainapp for posix_sitl_test to be the gtest main and run in the builds
  - add a snapdragon build to travis-ci and if possible get hardware connected to auto run these tests
  - auto test px4fmu hardware? I tried a Numato USB solid state relay, but the pixhawk wouldn't boot due to the voltage drop

 - testing for PX4 Matrix, ecl, DriverFramework should be run on each platform
 - PX4/DriverFramework to gtest? These tests don't currently cause the build to fail
 - resurrect https://github.com/PX4/fw-test-harness?
 - integration testing
   - HIL and SITL missions testing with at least one of each major airframe type
   - also need a way to test manual modes, failsafes, etc
 - once this settles update dev.px4.io documentation and start encouraging developers to add testing alongside new work

dagar, 2016-05-14:
It looks like getting gtest to build on Nuttx will take some effort.

bkueng, 2016-05-16:
Those are definitely good ideas. The most urgent one I see is to get snapdragon into travis. Snapdragon often breaks for various reasons, so this would make life for snapdragon devels easier.

A point I want to mention is performance: tests that run on PR requests should execute fast, otherwise it slows down our whole PR workflow. We can add more intensive tests that run eg every night on all branches.

I will probably be involved in HIL testing in the near future.

AndreasAntener, 2016-05-18:
Sounds good. Just letting you know that I'm working on SITL mission testing. The main issue there is still reporting failures in a way you can understand what went wrong, or at least being able to run the test locally with a one liner would be great. The less important issue (but will become important) is simulation speed. On decent hardware we can let Gazebo run at multiple times of realtime, but our simulator needs to be coupled to simulation time (not sure where that stands, I remember someone requested being able to pause via Gazebo not long ago).

dagar, 2016-05-19:
Could we pull logmuncher or equivalent into a script in Tools? For a failure maybe we could annotate the plots with when the failure occurred. We could keep the logmuncher output of the last successful run of that particular test on s3 for comparison.

julianoes, 2016-11-22:
Closing, @dagar if there is anything missing of the points, please add a separate issue.

closed 2016-11-22 by julianoes