PX4/PX4-Autopilot #4534: Descend mode when GPS fails needs checking

julianoes, 2016-05-13:
The case where RC and GPS, both have failed, needs to be checked. It should go into open-loop descend mode, however, it doesn't seem to execute it correctly and just drifts off.

Additionally, it can't auto-disarm after landing because the landing detector doesn't work when GPS is lost.

tubeme, 2016-05-16:
Probably this is the reason I had almost terrible accident with a friend of mine. My VTOL flipped after rough landing, far away in pretty deep grass. So in the rush I did not disarm the machine thinking that auto disarm will trigger. My friend went to pick up the VTOL, He saw that the machine is upside down spinning the props on idle. Once he picked up the machine it gave full throttle in his hands. It was almost fatal. I heard the props but was away from my RC. Finally I was able to disarm. But THIS IS DANGEROUS! It happened probably because when flipped the VTOL did not have GPS reception. I could not find the reason until now and could not explain how is it possible that the auto disarm did not trigger for 3 minutes obviously on the ground flipped over.

julianoes, 2016-05-17:
@tubeme I don't think that your issue is directly connected to the GPS fail (unless a log proves it). If it flips, the impact is likely to throw off the position estimator, so it might just not have detected the landing.
I agree that it would be good if an upside down landing would get detected properly, it's something to chase up in a separate issue.

However, I would argue that the "danger" in this case clearly came from approaching and picking up a vehicle when it is still armed.

julianoes, 2016-05-17:
@tubeme I created an issue for your case: #4558.

tubeme, 2016-05-17:
@julianoes We are preparing an end user product, so I'm expecting a lot of inconsistent behavior around this machines and many other machines based on PX4. The situation I got in proves that even experienced user like me could do the mistake of not disarming... Partly because I thought the VTOL is a total crash and the batteries are disconnected which is the usual case... The user that got the VTOL had no idea about what is quad-copter and how it works neither for the idle speed etc. So he just picked it up by not knowing what will follow. I expect a lot of similar situations the more user choose PX4, so there should be protection.  I had an incident two weeks ago with our VTOL and had my tow fingers almost chopped by the props, with blood etc. These things just can make a mess.

tubeme, 2016-05-18:
@julianoes Thanks for the new issue. Safety will distinguish the future players in the robot and drone industry.

julianoes, 2016-05-18:
@tubeme: Sorry, don't get me wrong. I totally agree that there should be protection (that's why I created thr separate issue) and I'm sorry to hear about your accident. This said, it's not always easily possible to detect failures without creating risks to detect false positives.

jcooper21, 2016-05-27:
@tubeme would the solution to your incident be crash detection and not land detection?  With APM any mild landing auto disarms because I have the crash detection set very low because I once had a similar issue when a plane went down and a farmer found it and picked it up.

julianoes, 2016-07-13:
Fixed in #4559.

closed 2016-07-13 by julianoes