PX4/PX4-Autopilot #4536: SITL ekf2 not starting

julianoes, 2016-05-13:
Seems like something went wrong with the switch to LPE:

```
make posix_sitl_ekf2 gazebo
...
rc_script: posix-configs/SITL/init/rcS_ekf2
debugger: none
program: gazebo
model: none
build_path: /home/julianoes/src/Firmware/build_posix_sitl_ekf2
empty model, setting iris as default
make[5]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[ 10%] Built target hello_world
[ 64%] Built target mav_msgs
[ 68%] Built target gazebo_opticalFlow_plugin
[ 72%] Built target LiftDragPlugin
[ 77%] Built target gazebo_lidar_plugin
[ 81%] Built target rotors_gazebo_controller_interface
[ 89%] Built target rotors_gazebo_imu_plugin
[ 89%] Built target rotors_gazebo_motor_model
[ 97%] Built target rotors_gazebo_multirotor_base_plugin
[100%] Built target rotors_gazebo_mavlink_interface

Built target sdf
WARN  Error opening file: ../../../../posix-configs/SITL/init/rcS_ekf2_gazebo_iris
Restoring terminal
[100%] Built target gazebo
```

LorenzMeier, 2016-05-13:
I don't think this is switch related. The config was just unmaintained.

LorenzMeier, 2016-05-19:
Thanks for fixing it!

closed 2016-05-19 by LorenzMeier