PX4/PX4-Autopilot #4542: A bug to report and a question to ask about the snapdragon_rc_pwm

SamChenzx, 2016-05-14:
I'm working on snapdragon flight these days. When I connect my snapdragon flight with a 3DR GPS/MAG module and start PX4 flight stack, I can arm and takeoff it via command line. The UART output file for snapdragon flight suppose to be snapdragon_rc_pwm.cpp.
In this source file I find that the baud rate as configured in initialise_uart() API is B921600. The problem is that if I modify the baud rate to B115200, the truly output baud rate is still keeping 921600. 
The question is I got the UART output data, instead of the PWM duty data, the data I got is actuator_controls->controls data, wander what's the meaning here?

julianoes, 2016-05-14:
Can you share your changes on a branch/PR/patch? Without looking at code, I can't really judge what's wrong.

I don't understand your question. Do you want PWM duty cycles for ESCs or controls in roll/pitch/yaw/thrust from -1 to 1?

SamChenzx, 2016-05-14:
I changed nothing but only modified B921600 to B115200.
I want PWM duty cycles for ESCs directly, is there anything else should be
changed?

julianoes, 2016-05-14:
Well, on the Snapdragon, uart_esc.cpp is running, you have to change the baudrate there as well. As far as I know, it should already send PWM signals (over mavlink).

SamChenzx, 2016-05-14:
Thanks for your remind, now I got the PWM signals, so the bug seems in the
Qtcreator on Ubuntu Linux. I use it as my code editor but can only
find snapdragon_rc_pwm
in the project file tree uart_esc in the file system but not included in
the project.

SamChenzx, 2016-05-20:
@julianoes A little bit more issues. I want to use RC to control snapdragon flight now. Reviewing the snapdragon_rc_pwm.cpp, I find the below code in task_main function:

       // check if we have new rc data, if yes send it to snapdragon
		bool rc_updated = false;
		orb_check(_rc_sub, &rc_updated);
		if (rc_updated) {
			orb_copy(ORB_ID(input_rc), _rc_sub, &_rc);
			// send mavlink message
			send_rc_mavlink();
		}

orb_copy() reads an input_rc_s structure with the format:

struct input_rc_s {
#endif
	uint64_t timestamp; // required for logger
	uint64_t timestamp_publication;
	uint64_t timestamp_last_signal;
	uint32_t channel_count;
	int32_t rssi;
	uint16_t rc_lost_frame_count;
	uint16_t rc_total_frame_count;
	uint16_t rc_ppm_frame_length;
	uint16_t values[18];
	bool rc_failsafe;
	bool rc_lost;
	uint8_t input_source;
	uint8_t _padding0[3]; // required for logger

#ifdef __cplusplus
	static const uint8_t RC_INPUT_SOURCE_UNKNOWN = 0;
	static const uint8_t RC_INPUT_SOURCE_PX4FMU_PPM = 1;
	static const uint8_t RC_INPUT_SOURCE_PX4IO_PPM = 2;
	static const uint8_t RC_INPUT_SOURCE_PX4IO_SPEKTRUM = 3;
	static const uint8_t RC_INPUT_SOURCE_PX4IO_SBUS = 4;
	static const uint8_t RC_INPUT_SOURCE_PX4IO_ST24 = 5;
	static const uint8_t RC_INPUT_SOURCE_MAVLINK = 6;
	static const uint8_t RC_INPUT_SOURCE_QURT = 7;
	static const uint8_t RC_INPUT_SOURCE_PX4FMU_SPEKTRUM = 8;
	static const uint8_t RC_INPUT_SOURCE_PX4FMU_SBUS = 9;
	static const uint8_t RC_INPUT_SOURCE_PX4FMU_ST24 = 10;
	static const uint8_t RC_INPUT_SOURCE_PX4FMU_SUMD = 11;
	static const uint8_t RC_INPUT_SOURCE_PX4FMU_DSM = 12;
	static const uint8_t RC_INPUT_SOURCE_PX4IO_SUMD = 13;
	static const uint8_t RC_INPUT_MAX_CHANNELS = 18;

#endif
};

First question is: If I use a third part radio receiver connecting to snapdragon flight's J15 RX pin, should I modify my radio receiver output as the input_rc_s structure or other structure like __mavlink_rc_channels_t so it can be received by J15's RX pin and will used by snapdragon?

Second question is:  send_rc_mavlink() function seems send RC signals output by J15 TX pin, and should I connect J15's TX pin to J13's RX pin, as I find the handle_message() function in [uart_esc.cpp](https://github.com/PX4/Firmware/blob/master/src/drivers/uart_esc/uart_esc.cpp#L317) seems to handle RC signals and advertise it by: 
`
else {
		_rc_pub = orb_advertise(ORB_ID(input_rc), &_rc);
	}
`

Inaddition, as dig into  handle_message() in snapdragon_rc_pwm.cpp, I found set_pwm_output() function:

			if (!isnan(actuator_controls->controls[i])) {
				long unsigned pwm = actuator_controls->controls[i];
				int ret = ::ioctl(_pwm_fd, PWM_SERVO_SET(i), pwm);
				if (ret != OK) {
					PX4_ERR("PWM_SERVO_SET(%d)", i);
				}
			}

So the third question is as it looks like output PWM signals directly to PWM_SERVO_SET, but where are those ports on the snapdragon flight hardware, and should I connect the TX of J13 to the RX of J15?

Quite confused by these issues, many thanks if you can help to clarify them.

julianoes, 2016-06-08:
@SamChenzx: Sorry that I didn't respond earlier here. Your confusion comes from the fact that snapdragon_rc_pwm.cpp runs on a Pixhawk/Pixfalcon/Pixracer and does the input output for the Snapdragon. In order to use it, you set the SYS_AUTOSTART param to 20000 and the Pixhawk then acts as a passthrough for Snapdragon.

I'm assuming what you want is to have RC input without having to use another microcontroller. This should be possible when using a dsm receiver which is just serial. There is code around for this but I'm not sure what the state exactly is, I'd have to do some digging.

SamChenzx, 2016-06-08:
Thanks, now I'm clear about that.

julianoes, 2016-06-08:
Ok, do you have more questions that I didn't see here, or can we close the issue?

SamChenzx, 2016-06-08:
Close this issue please.

closed 2016-06-08 by julianoes