PX4/PX4-Autopilot #4544: Replay of ulog files

LorenzMeier, 2016-05-14:
@bkueng @dagar Now that we can log the whole system I would love to get going with an app able to replay a log into the system. That would not be too hard to generate and extremely powerful.

dagar, 2016-05-14:
Interesting idea. Do we only replay "input" messages? gps, sensors, manual_input, etc.

LorenzMeier, 2016-05-14:
We replay by default everything, but we should have a blacklist so that you can disable the messages of the module you're intending to test.

kd0aij, 2016-05-14:
What datarate will be required to log everything, and will we be limiting
the logging rate for high-rate sensors? We'll need to increase the limit on
file descriptors to log all topics, and perhaps allocate more stack space.

bkueng, 2016-07-11:
done in https://github.com/PX4/Firmware/pull/4817

closed 2016-07-11 by bkueng

junhohan, 2017-04-19:
A Great feature! @bkueng saved me!!