PX4/PX4-Autopilot #4546: LPE: Takeoff command

ecmnet, 2016-05-14:
Switching to state MAIN_STATE_AUTO_TAKEOFF requires home position set as well as a valid global position. LPE (with GPS off) however publishes global position only, when a certain altitude above ground is reached. As a consequence, TakeOff does not work with LPE. @jgoppert: Any idea to overcome this?

jgoppert, 2016-05-14:
There is no valid position data when you are on the ground with a short vehicle because the flow is out of range. So you need to launch in manual, and then flip it to auto. Maybe would could create some takeoff command that says just go into altitude mode and climb, but that seems dangerous.

ecmnet, 2016-05-14:
@jgoppert I agree, TakeOff without any position control seems to be dangerous. What about if GPS is enabled in LPE? The same behavior?

jgoppert, 2016-05-14:
@ecmnet If GPS is enabled you can do auto take-off. There is no height requirement, there is only the requirement that you have position data.

ecmnet, 2016-05-14:
@jgoppert I don't think so. As far as I found out now, global position is only set, when flow is initialized in LPE, even with GPS enabled. Also home-position is only set with valid flow.

jgoppert, 2016-05-14:
That's odd, I've done auto take offs with GPS. Maybe it is a recent issue or something I missed when you are running both GPS and flow.

ecmnet, 2016-05-15:
@jgoppert I just figured out, that my GPS was too bad, which did not allow global position being published in landed state. As LPE publishes global position, if any of the XY sensors is available, it is now clear, why home position was set after takeoff only in my case. Flow was working then, allowing global position to be published and taken as home position.

closed 2016-05-15 by ecmnet