PX4/PX4-Autopilot #4550: External Mag Rotation

dagar, 2016-05-16:
With a fresh pixhawk (all params reset) and an external mag there doesn't seem to be a way to set the rotation in QGC sensor calibration. Unless I've missed something it appears that you're only prompted for the rotation if CAL_MAGi_ID is set and CAL_MAGi_ROT != -1, but those aren't set until you calibrate.

https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/SensorsComponent.qml#L109

One way to fix this would be properly filling in CAL_MAG*_ID in sensors. It gets a bit more complicated with the possibility of external mags coming and going and not wanting to throw away calibration until you have to.
https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L1386

DonLakeFlyer, 2016-05-16:
You should be prompted for rotations with a side dialog after cal completes.

dagar, 2016-05-16:
Ha, I totally missed that. Thanks Don

closed 2016-05-16 by dagar