PX4/PX4-Autopilot #4552: VTOL in MC mode gets confused about what is level flight

dzett, 2016-05-16:
Hi there,
There currently are three independent reports of VTOLs flipping and maintaining + or -90deg pitch shortly after takeoff in MC mode and remaining uncontrollable in a meaningful way, see http://discuss.px4.io/t/caipi-vtol-pitch-issues-with-master-firmware-plane-goes-level-in-mc-mode incl. logs. At least my Caipi tailsitter build did not show this with a master from end of March/early April where I was able to remain level and keep control during a short test flight, something I have since failed (with unmodified hardware). Seems not to be limited to the Tailsitter config.

@tumbili promised to check against the reference Caipi, just wanted to get this tracked as an official issue since it apparently is not just limited to my build.

Thanks,
Daniel

tumbili, 2016-05-16:
@dzett Hi, I just flashed my lab tailsitter with latest master and everything seems to be working.
I can upload a log for you to compare your parameters.
Will do that in a sec.

tumbili, 2016-05-16:
@dzett Here is the log: http://logs.uaventure.com/view/E5H6ezBC3WYTusjaCTN7La

tumbili, 2016-05-16:
@dzett Here is what I think:
1) Tuning of the pitch axis might not be sufficient. In my last flight my pitchrate P gain was at 0.5 compared to the 0.3 you were using. We possibly have to adjust the default gains for the Caipirinha because several things have changed over the last year and we are not able to push the gains higher.
2) The default elevons of the Caipi are definitely too small. My hopes were that we would be ok with the optimal recovery strategy but even then it's just not enough. I can fly ok in rooms and outside if there is really now wind. But from my experience wind will kill you with this airframe. I have other tailsitter airframes which have been designed to hover and they perform much better in wind, e.g. they are always able to recover the error in pitch.

dzett, 2016-05-16:
@tumbili, thanks, then I will try with adjusted tuning according to your settings and a fresh master but unfortunately have to wait until next weekend. 
I just found it odd that even fully pulling back pitch towards myself during takeoff and while still upright never caused the plane to tip towards me (which it would before).
Also, the other users have different frames (like generator on discuss.px4.io with larger elevons and Rolf at fpv-treff who even has a tilt rotor setup) and they also reported the pitching either forward or backward... Or is this maybe also trimming related, some special considerations there for tailsitters?

Did you have a chance to look into the other points generator reported in the thread discuss.px4.io? Maybe it would be helpful if you could shed some light on what could be causing this and what the intended behavior is there (maybe some "education" for us needed there beyond the scope of this issue).

Will close the issue once I were able to retest if that is OK for you.

dzett, 2016-05-16:
@tumbili two more questions if I may :-) 

- The pitch tracking in your log indeed looks better. Is that with the standard elevons or already modified?

- I compared the parameters in our logs and besides tuning and calibration (and some new params that seem to have been introduced the last days eg around EKF2) and a bit more aggressive values for pos control velocities they seem identical. One difference is the circuit breaker for airspeed being set for mine - could this be causing the issues as there are several places in the vtol code that make use of the airspeed value?

tumbili, 2016-05-16:
@dzett Let me set the circuit braker and see if this is the problem.

dzett, 2016-05-17:
@tumbili thanks for your offer and your help, but on second thought it is unlikely that it is the airspeed circuit breaker because I had also set it during my earlier flight, so you can save the time. Once I am finally back I will try the newest master and improved PIDs.

dzett, 2016-05-22:
Can be closed, thanks for your help. As reported also in the referenced thread, seems tailsitter performance is normal and the issues seen are more related to tuning and limitations. I have also been able to reproduce the performance I had been seeing before with the older master while inside. Improving field conditions will require additional measures on the side of the plane.

Will follow advice by @tumbili and get larger elevons installed to further improve hover Performance.

closed 2016-05-22 by dzett