PX4/PX4-Autopilot #4556: Transform roll/pitch commands with current yaw instead of setpoint

AndreasAntener, 2016-05-17:
@tumbili @LorenzMeier 
VTOLs really suffer from this currently, would be great if we can get to the bottom of it soon.

For reference, a similar previous attempt: https://github.com/PX4/Firmware/pull/3397

tubeme, 2016-05-17:
There is something inadequate in the YAW handling in the PX4 for the VTOL. I dont know what, it might be the issue here and the #3397 as well.

Benchmark is our simple KK2.0 build, which handles the YAW much much more adequate in hover ie quad.

If I want to really have good YAW handling in all modes with PX4 I need to crank so much the PID's that the osculations are obvious. When I tune the frame to be steady and without osculations, then the YAW is very very mellow and soft and not strong enough. Kind of like it is drunk person not very sharp in the responses especially in windy weather. Using POS Ctl in windy weather in a hover, becomes kind of challenging for the machine.

Also we've had a corner case, which I will explain in another new issue in detail, in which If we are in Stabilized flight and we make a back transition in a 20-30% throttle,  the VTOL starts sliding like it is with skates on ice. And most noticeably it starts spinning (YAW) left to right or sometimes 180. And in this condition even if we apply some 60 - 70% throttle to recover the recovery is very slow and sometimes fatal.

Of coerce the reason for this is, we don't give enough throttle to the quad and after the transition the quad spins very slow, the VTOL continues in a very steady and fast descend with a fast forward motion from the inertia of the FW flight and with yaw spin. If you delay the application of the throttle with just 2 seconds this condition is almost unrecoverable after that with the quad.

I don't know what it is but there is something with the Yaw.

LorenzMeier, 2016-05-17:
@tubeme Have you yaw feedforward (FF) set? Sounds like you're missing it.

tubeme, 2016-05-17:
@LorenzMeier Yes the FF is not enough now when you pointed out. Will try more FF and will report.

Though how's the back transition YAW problem related to FF gain? Isn't it that the FF gain is applied to my input form the RC?

AndreasAntener, 2016-05-17:
Oh, I thought there was already an issue for that, ref: https://github.com/PX4/Firmware/issues/3620

tumbili, 2016-05-18:
Fixed in #4564

PR #4564 (2016-05-19): WIP: Manual attitude setpoint for large heading errors
mc pos control: In manual mode calculate attitude setpoint such that it reflects the users intuition of roll and pitch for any given heading error.

TODO:
- [x] extensive testing in SITL
- [x] testing on real multicopter
- [ ] review
- [x] add comments to code

Fixes #4556
Fixes #3620
closed 2016-05-18 by tumbili