PX4/PX4-Autopilot #4558: Flipped landing detection

julianoes, 2016-05-17:
It would be nice if the land detector would robustly detect when the vehicle has landed upside down or just flipped when landing.

jcooper21, 2016-05-27:
It would be beneficial to set the servos to neutral upon landing in case your upside down with a fixed wing.

dagar, 2017-04-12:
Still interested in this? Is the point to use that information to kill the motors?

julianoes, 2017-04-12:
Yes, this would still be a good thing to have.

TSC21, 2017-07-25:
@julianoes maybe we can discuss this in a call or so to think in a proper approach to this? Really interested on it also.

TSC21, 2018-02-21:
I consider this still relevant and worth discussion.

bresch, 2022-08-15:
Flip detection is covered by roll/pitch failure cases in FailureDetector

closed 2022-08-15 by bresch