PX4/PX4-Autopilot #4560: reduce ifdef CONFIG_ARCH_BOARD_XXXXX in code base

davids5, 2016-05-17:
1) Consider using board_config.h to define components and buses

 For example code the says
   ```#if defined(CONFIG_ARCH_BOARD_XXXXX)``` becomes ```#if defined(HAVE_MPU6500)```
see https://github.com/PX4/Firmware/blob/master/src/drivers/drv_sensor.h#L75

2) Use the same mechanism that is afforded by xxx_timer_config.c to define io [resets] (https://github.com/PX4/Firmware/blob/master/src/drivers/px4fmu/fmu.cpp#L2170) in fmu.cpp and IO [defines] (https://github.com/PX4/Firmware/blob/master/src/drivers/px4fmu/fmu.cpp#L274) .

davids5, 2016-05-25:
@LorenzMeier - here is the issue and the wip

See master_micro_hal_device Wip - DO NOT MERGE https://github.com/PX4/Firmware/pull/4641

There are #defines of  PX4_I2C_OBDEV_{HMC5883} in board_configs.h that do not have that device on board but need the driver.
 
A board has a bus {spi|i2c}, a device is i2c and/or spi but it is wired to only one bus one way.
It can be on board or off board
For i2c the address is either the default - or there are strapping option and it can be strapped to an address.
So if a device is on board it is tied to one of the i2c or spi buses.

I think - if it is on i2c and has address strapping options pins that are strapped to the non defailt, then the board_config shoud define the stapped address:
PX4_I2C_OBDEV_HMC5883_ADDRESS the driver should use the device default address if none is provided by the board_config.h. 

Then the i2c drivers should:

For PX4_I2C_BUS_EXPANSION and the default address or the cmd line options to select the address. 

For PX4_I2C_BUS_ONBOARD if PX4_I2C_OBDEV_HMC5883_ADDRESS is not defined, then use the device default address.

for SPI device 
Define and an entry based on definition of PX4_SPIDEV_{HMC} and probe for it

dagar, 2017-04-11:
Update?

davids5, 2017-04-11:
@dagar - No I need time or help to implement.

closed 2018-02-21 by PX4BuildBot