PX4/PX4-Autopilot #4561: ver.c Cleanup

davids5, 2016-05-17:
1.  Rework https://github.com/PX4/Firmware/blob/master/src/systemcmds/ver/ver.c#L66 to work like the PX4_GIT_TAG_STR is generated from the last NuttX tag
2. Consolidate all the unique_id like mcu_unique_id to use boardctl(BOARDIOC_UNIQUEID) (produce the same format as uavcan requires with a wrapper to keep ver command format the same

davids5, 2016-12-09:
on nuttx_v3 need to expand support for new procs

dagar, 2016-12-13:
Add something for fmu-v3?

davids5, 2016-12-14:
As it is fmu-v3 does not have a BOARD_NAME it "inherits" fmu-v2 and uses is different Linker script. We may be able to  to undef BOARD_NAME and redefine it. Then ripple that to the startup script

LorenzMeier, 2017-04-10:
Done.

closed 2017-04-10 by LorenzMeier