PX4/PX4-Autopilot #4567: Generic 250 Voltage Parameters Wrong

jgoppert, 2016-05-18:
When gunning the motors I was able to drop the voltage 1-2 volts. This was triggering the battery alarm, so we need to figure out the correct setting for the voltage drop/ current parameter. 

BAT_V_LOAD_DROP	Voltage drop per cell on full throttle	0.07	0.5	0.3
This implicitely defines the internal resistance to maximum current ratio and assumes linearity. A good value to use is the difference between the 5C and 20-25C load.

So maybe we need to increase this from 0.3 to about 0.6 (0.3*3~1 volt/amp), which is above the maximum.

kd0aij, 2016-05-18:
@jgoppert I don't see any of those parameters being set in existing quad config files. I'm sure the drop under load is battery and vehicle specific too. My solution has been just to watch the voltage at my transmitter (or OSD) and ignore the PX4 warnings :)

jgoppert, 2016-05-18:
@kd0aij Granted, I think we should still add more reasonable defaults to the generic 250. At full throttle I went from 12.55 to 10 volts at times (on my OSD), no reason we can't fix that with better parameters. And you would hope that the voltage drop is linear with respect to current draw at least for similarly sized vehicles. I'll try to get some logs of it, so we can smooth it out. No one wants to be zipping along 30 seconds in and hear critical battery warning alarms.

kd0aij, 2016-05-18:
Agreed. I'll see if I can find good values from my S250 and QAV250. I note that the Vortex OSD shows a warning when remaining capacity drops below a certain point. Seems like the threshold may be 400mah

jgoppert, 2016-05-18:
@kd0aij yeah I think it should warn on extreme voltages or capacity past max ideally. Just in case you start flying with an empty battery.

LorenzMeier, 2016-12-28:
Closing as outdated.

closed 2016-12-28 by LorenzMeier