PX4/PX4-Autopilot #4570: PixRacer: shorten PWM ramp and arming delay

kd0aij, 2016-05-18:
The arm/disarm delay for PixRacer can be shortened considerably; the Blade Vortex seems to be <1 second. And the long PWM ramp is completely unnecessary. Worse, the throttle is inhibited till the ramp ends and users are throttling up during that period and getting a sudden high-throttle start.

LorenzMeier, 2016-05-19:
I would shorten the ramp for all systems to 1-1.5 seconds. Would you want to do a PR for that?

kd0aij, 2016-05-19:
I'll do some testing to determine whether the ramp is necessary. Was it intended for safety reasons or was it necessary at some point to accommodate a bad ESC? I've never found it necessary to start an ESC with a ramp.

LorenzMeier, 2016-05-19:
If you don't have your ESCs calibrated it will lead to a flip if you don't use a ramp.

pkocmoud, 2016-05-19:
Why not make this a param? Problem solved. Default to 1.5 seconds. Mark can change to 0 or anyone else wanting its removal.

kd0aij, 2016-05-19:
@LorenzMeier It looks like using params for this would be a PITA, since the ramp is done in the IO coprocessor, and it doesn't have access to parameters.

LorenzMeier, 2016-05-19:
Let's just shorten it. Half a second to a second might just do it.

kd0aij, 2016-05-25:
addressed by #4594 and #4579

closed 2016-05-25 by kd0aij