PX4/PX4-Autopilot #4571: timestamp in msg files

marianobianchi, 2016-05-18:
Since commit 76387b1693f234602e6c354115fbcac25b33baab all .msg files stop having a timestamp value. But code from nodes still uses it so when i try to compile using catkin_make i get these errors:

```
catkin_ws/src/px4/src/platforms/ros/nodes/attitude_estimator/attitude_estimator.cpp: In member function ‘void AttitudeEstimator::ModelStatesCallback(const ModelStatesConstPtr&)’:
catkin_ws/src/px4/src/platforms/ros/nodes/attitude_estimator/attitude_estimator.cpp:98:12: error: ‘px4::vehicle_attitude {aka struct px4::vehicle_attitude_<std::allocator<void> >}’ has no member named ‘timestamp’
  msg_v_att.timestamp = px4::get_time_micros();
            ^
catkin_ws/src/px4/src/platforms/ros/nodes/attitude_estimator/attitude_estimator.cpp: In member function ‘void AttitudeEstimator::ImuCallback(const ImuConstPtr&)’:
catkin_ws/src/px4/src/platforms/ros/nodes/attitude_estimator/attitude_estimator.cpp:141:12: error: ‘px4::vehicle_attitude {aka struct px4::vehicle_attitude_<std::allocator<void> >}’ has no member named ‘timestamp’
  msg_v_att.timestamp = px4::get_time_micros();
            ^
Linking CXX executable catkin_ws/devel/lib/px4/demo_offboard_position_setpoints
catkin_ws/src/px4/src/platforms/ros/nodes/commander/commander.cpp: In member function ‘void Commander::ManualControlInputCallback(const manual_control_setpointConstPtr&)’:
catkin_ws/src/px4/src/platforms/ros/nodes/commander/commander.cpp:72:28: error: ‘px4::vehicle_control_mode {aka struct px4::vehicle_control_mode_<std::allocator<void> >}’ has no member named ‘timestamp’
  _msg_vehicle_control_mode.timestamp = px4::get_time_micros();
                            ^
catkin_ws/src/px4/src/platforms/ros/nodes/commander/commander.cpp:76:22: error: ‘px4::actuator_armed {aka struct px4::actuator_armed_<std::allocator<void> >}’ has no member named ‘timestamp’
  _msg_actuator_armed.timestamp = px4::get_time_micros();
                      ^
catkin_ws/src/px4/src/platforms/ros/nodes/commander/commander.cpp:96:22: error: ‘px4::vehicle_status {aka struct px4::vehicle_status_<std::allocator<void> >}’ has no member named ‘timestamp’
  _msg_vehicle_status.timestamp = px4::get_time_micros();
                      ^
catkin_ws/src/px4/src/platforms/ros/nodes/commander/commander.cpp:106:53: error: ‘px4::parameter_update {aka struct px4::parameter_update_<std::allocator<void> >}’ has no member named ‘timestamp’
  if (px4::get_time_micros() - _msg_parameter_update.timestamp > 1e6) {
                                                     ^
catkin_ws/src/px4/src/platforms/ros/nodes/commander/commander.cpp:107:25: error: ‘px4::parameter_update {aka struct px4::parameter_update_<std::allocator<void> >}’ has no member named ‘timestamp’
   _msg_parameter_update.timestamp = px4::get_time_micros();
                         ^
catkin_ws/src/px4/src/platforms/ros/nodes/commander/commander.cpp: In member function ‘void Commander::OffboardControlModeCallback(const offboard_control_modeConstPtr&)’:
catkin_ws/src/px4/src/platforms/ros/nodes/commander/commander.cpp:214:23: error: ‘px4::vehicle_status {aka struct px4::vehicle_status_<std::allocator<void> >}’ has no member named ‘timestamp’
   _msg_vehicle_status.timestamp = px4::get_time_micros();
                       ^
catkin_ws/src/px4/src/platforms/ros/nodes/commander/commander.cpp:221:23: error: ‘px4::actuator_armed {aka struct px4::actuator_armed_<std::allocator<void> >}’ has no member named ‘timestamp’
   _msg_actuator_armed.timestamp = px4::get_time_micros();
                       ^
catkin_ws/src/px4/src/platforms/ros/nodes/commander/commander.cpp:223:29: error: ‘px4::vehicle_control_mode {aka struct px4::vehicle_control_mode_<std::allocator<void> >}’ has no member named ‘timestamp’
   _msg_vehicle_control_mode.timestamp = px4::get_time_micros();
```

LorenzMeier, 2016-05-18:
The timestamp is automatically added since that change: http://dev.px4.io/advanced-uorb.html#adding-a-new-topic

@bkueng That might be missing from our ROS port. @marianobianchi Are you actually using the ROS port after all? Its incomplete and undocumented.

marianobianchi, 2016-05-18:
Thanks for your answer. I am working on a project that have a little code for using multiple UAVs and I'm trying to run that code with the latest firmware (latest SITL). I thought ROS integration was still working as always but now that you say this I might have started off on the wrong foot. I don't know what do you mean with "ROS port" but I think you mean that is not recommended to include px4 firmware as a source of a catkin workspace, is it?

I will try to follow [these](http://dev.px4.io/simulation-ros-interface.html#launching-gazebo-with-ros-wrappers) instructions and see how it works.

bkueng, 2016-05-19:
Yes I think the catkin workspace does not work, but the ROS wrapper does.

closed 2016-05-19 by LorenzMeier