PX4/PX4-Autopilot #4573: Problematic cmake target.

intijk, 2016-05-19:
When I do:

`make gazebo_build`

it gives:

```
...
...
make[1]: *** No targets specified and no makefile found.  Stop.
```

I did some tracking and found in **Makefile**
```

gazebo_build:
        mkdir -p build_gazebo
        if [ ! -e $(PWD)/build_gazebo/CMakeCache.txt ];then cd build_gazebo && cmake -Wno-dev -G$(PX4_CMAKE_GENERATOR) $(PWD)/Tools/sitl_gazebo; fi
        cd build_gazebo && $(PX4_MAKE) $(PX4_MAKE_ARGS)
        cd build_gazebo && $(PX4_MAKE) $(PX4_MAKE_ARGS) sdf

```

it writes the build files to **Tools/sitl_gazebo**, but went to **build_gazebo** to perform the make.

Should change

        cd build_gazebo && $(PX4_MAKE) $(PX4_MAKE_ARGS)
        cd build_gazebo && $(PX4_MAKE) $(PX4_MAKE_ARGS) sdf

to

        cd Tool/sitl_gazebo && $(PX4_MAKE) $(PX4_MAKE_ARGS)
        cd Tool/sitl_gazebo && $(PX4_MAKE) $(PX4_MAKE_ARGS) sdf

This problem cause all make command in this page does not work:

http://dev.px4.io/simulation-gazebo.html

Please check it.

dagar, 2016-05-19:
Cmake writes the build files to the current directory and you specify the source directory or an existing build directory. The gazebo targets (eg `make posix_sitl_default gazebo`) are currently working for me. I'm guessing you have leftover build files from before the build_gazebo directory was moved.

Can you try deleting build_gazebo, and cleaning Tools/sitl_gazebo (git clean -ff -x -d)? Then try `make posix_sitl_default gazebo`.

intijk, 2016-05-19:
I carefully checked my clean, I found **git clean -ff -x -d** cannot really clean the folder (Tools/sitl_gazebo still have files which cannot be remove by git clean ).

I tried to clone the repo again, now it works, thank you :)

mhkabir, 2016-05-19:
@intijk Next time, run ```make submodulesclean```

closed 2016-05-19 by mhkabir

dagar, 2016-05-19:
I'm glad it's working. 
I'm surprised git clean didn't work. Did you run it from Tools/sitl_gazebo? It's a submodule so running it in a directory above won't clean within sitl_gazebo.

I've updated `make clean` to also clean Tools/sitl_gazebo.

intijk, 2016-05-19:
As @mhkabir mentioned.

**make submodulesclean** did the work :)