PX4/PX4-Autopilot #4575: sitl01.dronetest.io failures

dagar, 2016-05-19:
@AndreasAntener @LorenzMeier 
The PX4 Firmware POSIX SITL jobs seem to fail failure often due to email server problems. Maybe disable email notifications?

http://sitl01.dronetest.io/job/PX4%20Firmware%20POSIX%20SITL%20PRs/542/console

Sending e-mails to: andreas@antener.name lm@qgroundcontrol.org
ERROR: Could not connect to SMTP host: localhost, port: 25
javax.mail.MessagingException: Could not connect to SMTP host: localhost, port: 25;
  nested exception is:
	java.net.ConnectException: Connection refused
	at com.sun.mail.smtp.SMTPTransport.openServer(SMTPTransport.java:1934)
	at com.sun.mail.smtp.SMTPTransport.protocolConnect(SMTPTransport.java:638)
	at javax.mail.Service.connect(Service.java:295)
	at javax.mail.Service.connect(Service.java:176)
	at javax.mail.Service.connect(Service.java:125)
	at javax.mail.Transport.send0(Transport.java:194)
	at javax.mail.Transport.send(Transport.java:124)
	at hudson.tasks.MailSender.run(MailSender.java:126)
	at hudson.tasks.Mailer.perform(Mailer.java:141)
	at hudson.tasks.BuildStepCompatibilityLayer.perform(BuildStepCompatibilityLayer.java:74)
	at hudson.tasks.BuildStepMonitor$1.perform(BuildStepMonitor.java:20)
	at hudson.model.AbstractBuild$AbstractBuildExecution.perform(AbstractBuild.java:761)
	at hudson.model.AbstractBuild$AbstractBuildExecution.performAllBuildSteps(AbstractBuild.java:721)
	at hudson.model.Build$BuildExecution.post2(Build.java:183)
	at hudson.model.AbstractBuild$AbstractBuildExecution.post(AbstractBuild.java:670)
	at hudson.model.Run.execute(Run.java:1743)
	at hudson.model.FreeStyleBuild.run(FreeStyleBuild.java:43)
	at hudson.model.ResourceController.execute(ResourceController.java:89)
	at hudson.model.Executor.run(Executor.java:240)
Caused by: java.net.ConnectException: Connection refused
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:339)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:200)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:182)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:579)
	at com.sun.mail.util.SocketFetcher.createSocket(SocketFetcher.java:286)
	at com.sun.mail.util.SocketFetcher.getSocket(SocketFetcher.java:231)
	at com.sun.mail.smtp.SMTPTransport.openServer(SMTPTransport.java:1900)
	... 18 more

AndreasAntener, 2016-05-19:
That's an issue but it's not the problem why the jobs fail (#4424).

dagar, 2016-05-19:
I completely missed that. Is it only trying to email on a test failure? If that's the case then just ignore this.

AndreasAntener, 2016-05-19:
Yep. Ah that's ok, we should fix it this too ;)

closed 2016-11-08 by dagar