PX4/PX4-Autopilot #4578: mavros_posix_sitl launch not connecting correctly

marianobianchi, 2016-05-19:
When running `roslaunch px4 mavros_posix_sitl.launch` I don't get a connection to work correctly from ROS. I checked what `/diagnostics` topic says and got this:

```
header: 
  seq: 1069
  stamp: 
    secs: 1066
    nsecs: 500000000
  frame_id: ''
status: 
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: udp://:14567@localhost:14557
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 247
      - 
        key: Rx total bytes:
        value: 0
      - 
        key: Tx total bytes:
        value: 295407
      - 
        key: Rx speed:
        value: 0.000000
      - 
        key: Tx speed:
        value: 277.000000
  - 
    level: 0
    name: mavros: GCS bridge
    message: connected
    hardware_id: udp://:14567@localhost:14557
    values: 
      - 
        key: Received packets:
        value: 0
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 0
      - 
        key: Tx sequence number:
        value: 0
      - 
        key: Rx total bytes:
        value: 0
      - 
        key: Tx total bytes:
        value: 0
      - 
        key: Rx speed:
        value: 0.000000
      - 
        key: Tx speed:
        value: 0.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: udp://:14567@localhost:14557
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 2
    name: mavros: Heartbeat
    message: No events recorded.
    hardware_id: udp://:14567@localhost:14557
    values: 
      - 
        key: Heartbeats since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Vehicle type
        value: Generic
      - 
        key: Autopilot type
        value: Generic
      - 
        key: Mode
        value: ''
      - 
        key: System status
        value: Uninit
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: udp://:14567@localhost:14557
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: udp://:14567@localhost:14557
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
  - 
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: udp://:14567@localhost:14557
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000
---
```

I checked what things mavros_posix_sitl.launch started and I find that it is connecting to the wrong udp port. mavros_posix_sitl.launch has this configuration for mavros:

```
<include file="$(find px4)/launch/mavros.launch">
    <arg name="ns" value="$(arg ns)"/>
    <arg name="fcu_url" value="udp://:14567@localhost:14557"/>
</include>
```

while in rcS_gazebo_iris (configuration used in posix_sitl.launch file, used to run run_sitl.sh script) the ports opened are 14557 and 14540:

```
mavlink start -u 14556 -r 4000000 -t 127.0.0.1
mavlink start -u 14557 -r 4000000 -m onboard -o 14540 -t 127.0.0.1
```

The problem is that if I change the mavros_posix_sitl.launch file to the correct udp ports I still get a FCU not connected message.

If I run `make posix_sitl_default gazebo` and in other console run `roslaunch mavros px4.launch fcu_url:="udp://localhost:14540@localhost:14557"` I get a FCU connection and can work from ROS without problems.

AndreasAntener, 2016-05-27:
Thanks for reporting this. It's fixed on master now. You need to start a fresh shell with the updated `source integrationtests/setup_gazebo_ros.bash $(pwd)` so the environment matches again.

closed 2016-05-27 by AndreasAntener

csorice147, 2016-10-22:
@AndreasAntener I get the same problem even though I'm using the updated master.
If I do `make posix_sitl_default gazebo` in 1 terminal and then do `roslaunch mavros px4.launch fcu_url:="udp://localhost:14540@localhost:14557"` it's all fine... 
Do you have any suggestions as to what might be the problem? The file `/integrationtests/setup_gazebo_ros.bash` does not seem to exist.

AndreasAntener, 2016-10-23:
@csorice147 things got moved a bit, execute `./Tools/setup_gazebo.bash` instead

julianoes, 2016-10-23:
The instructions should now be up-to-date, check out: http://dev.px4.io/simulation-ros-interface.html#launching-gazebo-with-ros-wrappers.

zubair123khan, 2018-06-17:
I am using "px4 mavros_posix_sitl.launch" and running a offboard mavros node for arming etc, but the node doesnt seems to have any impact. The node can be found here https://pastebin.com/mYP7QyS3.
I am not able to arm my IRIS with all this. Can any one tell me whats missing ?