PX4/PX4-Autopilot #4584: ROI Implementation and Gimbal Integration

thedevleon, 2016-05-19:
This is sort of 2 issues in one. If wanted, I can split this into two issues.

The issue is as following:
1. Nothing in regards to ROI is implemented. Neither the MAV_ROI enum, nor MAV_CMD_NAV_ROI or MAV_CMD_DO_SET_ROI are parsed by the commander.

2. Since those are not implemented, there is no current way to use the current vehicle ROI to control a gimbal (have it point to it).

Thus, I propose the following changes:
- [ ] Implement the `MAV_ROI` enum in the `vehicle_command` topic
- [ ] Create a new uORB topic for the vehicle ROI (named `vehicle_roi`)
- [ ] Implement the parsing of `MAV_CMD_NAV_ROI` and `MAV_CMD_DO_SET_ROI` in `commander.cpp` and the advertising of the ROI on the `vehicle_roi` topic
- [ ] Rename the current gimbal driver to something else like GimbalRC or RCGimbal<sup id="a1">[1](#f1)</sup>
- [ ] Create a proper gimbal driver that will subscribe to the `vehicle_roi` topic and send `MAV_CMD_DO_MOUNT_CONFIGURE` and `MAV_CMD_DO_MOUNT_CONTROL` commands to the gimbal<sup id="a2">[2](#f2)</sup>

If desired, I can start working on implementing those changes. Let me know if I missed anything or understood something wrong.

###### Footnotes

<b id="f1">1</b> Currently there is a gimbal driver, but it doesn't actually communicate with a gimbal via MavLink, but rather interprets the `MAV_CMD_DO_MOUNT_CONFIGURE` and `MAV_CMD_DO_MOUNT_CONTROL` commands and calculates pitch, roll and yaw and advertises them on the `actuator_controls_2` topic, to be used by a mixer, which then can be used to control a gimbal via it's RC inputs. [↩](#a1)

<b id="f2">2</b> Most gimbals now natively support MavLink (like the Storm32) and (as specified and intended by the MavLink specification) use the `MAV_CMD_DO_MOUNT_CONFIGURE` and `MAV_CMD_DO_MOUNT_CONTROL` commands in the `COMMAND_LONG` message to operate the gimbal. If a gimbal does not support MavLink, the old gimbal driver can still be used to interpret those message
[↩](#a2)

LorenzMeier, 2016-05-19:
@thedevleon Your proposal sounds excellent and correct regarding the vehicle architecture. It would be fantastic if you would be willing to start to work on it.

lvale, 2016-05-19:
@thedevleon Be aware that Mavlink parsing by gimbal controllers is not guaranteed, as some implement specific serial protocols (like AlexMos and also STorM32-both serial and mavlink)

thedevleon, 2016-05-19:
I agree @lvale. That's why I was saying that this is a 2 part issue.
The implementation, parsing and publishing of the ROI via uORB would be one thing, and the actual gimbal driver handling the communication (and limitations) would be the other :)

thedevleon, 2016-05-24:
There has been more development on this by me :) Please head over to #4624 for more info, as I changed things a bit as I went along.
It's already 90% done :D

dagar, 2016-11-08:
Implemented.

closed 2016-11-08 by dagar