PX4/PX4-Autopilot #4587: Standard VTOL pusher for pitch feature needs reconsideration

tumbili, 2016-05-20:
Since we have changed the creation of manual control setpoints we need to adjust the code which
uses the pusher thrust to accelerate forward. It will now not properly work if yaw errors are becoming larger.

tubeme, 2016-05-22:
You are talking about MC mode right?

tumbili, 2016-05-23:
Yes

tubeme, 2016-05-23:
This could be a separate VTOL mode and this way it will not intervene with the current classic implementation for the quad mode functioning . It is nice mode because it could make the control of the VTOL in hover in stronger weather better so it will be beneficial to the VTOL. 

Or if it is not separate mode then it could be turned on and off with parameter.

The problem with a separate mode is the transition. Because if we create the mode, let say  "Hybrid" then in what mode we fly in FW after the transition? Also it will be subset of MC mode so we need another switch to enable it etc.

Probably better with parameter.

closed 2016-06-02 by tumbili