PX4/PX4-Autopilot #4597: Tailsitter: reset yaw setpoint after back transition to the heading it had during FW flight

AndreasAntener, 2016-05-21:
(at least for manual)

pedro-sales, 2016-09-21:
@AndreasAntener Any updates or ideas on how to fix this?
I already modified the tailsitter code to transition using time when airspeed mode is disabled (no airspeed sensor) and it worked fine, but I am stuck at this yaw issue.

VTOLDavid, 2016-11-21:
I've found the same issue.  Any idea?? @tumbili @davidvor

VTOLDavid, 2016-11-30:
@AndreasAntener I'll put the label bug or critical... This can cause a crash.

AndreasAntener, 2016-11-30:
@VTOLDavid why do you think this can cause a crash? To me this is more a cosmetic bug, but maybe we're talking about a different thing?

VTOLDavid, 2016-11-30:
@AndreasAntener  This is what I've detected, I think it's the same problem:
When the aircraft switches from Fw mode to MC the multicopter actuator turns the aircraft towards heading 0. I supose that the reference heading is zeroed instad of set the current heading when the transition begins. It makes the aircraft turn violently towards north during transition from FW to MC.

If the MC controller is engaged when you still have vertical speed It can be dangerousspecially if your airframe has enough yaw autority with elevons.

closed 2018-02-21 by PX4BuildBot

VTOLDavid, 2018-02-21:
I'll reopen this, if it is not solved...