PX4/PX4-Autopilot #4598: Tailsitter: add ability to scale control outputs differently in MC and FW mode

AndreasAntener, 2016-05-21:
For example: tailsitters that need extreme control surface outputs in MC are difficult to fly manually in FW.

tumbili, 2016-05-22:
@AndreasAntener I don't quite understand this problem. The controllers in mc and fw and different so each one can be tuned for the appropriate flight regime.

AndreasAntener, 2016-05-25:
This is not about the PIDs, but straight output scaling when in full manual control in FW. In MC the controller obviously needs as much control surface throws as possible, but in FW for "normal" flying I might only need about half of the possible deflections. At the moment the pilot needs to make a mental switch during the transition and "scale" his stick input accordingly.

tumbili, 2016-05-25:
@AndreasAntener Ok, I wasn't thinking about manual flying. This is probably something we can do on the fw side (not VTOL specific) since it can be used for normal fixed wings as well. At the moment I think the only way to do it for fw is to change the scale in the mixer.

kd0aij, 2016-05-25:
I use a programmable TX to change "rates" based on flight mode. From my
point of view, duplication of this logic in the firmware simply adds
complexity to the overall system, but, of course, it enables the use of
"dumb" RC transmitters.

AndreasAntener, 2016-05-25:
Sure, that's of course what I ended up doing ;). But I think it's a very annoying thing to do. If your autopilot already knows your system, why should I have to set anything at all on my remote?

kd0aij, 2016-05-25:
Wouldn't there be new parameters equivalent to the ones you set in your TX? And wouldn't you have to set those instead?

AndreasAntener, 2016-05-25:
That's why we have vehicle configurations? Since I started using PX4 I got down from many model configurations on my RC to 1, I'd like to keep it that way :)

kd0aij, 2016-05-25:
True, but *somebody* has to create the vehicle config, and I was thinking like a DIYer creating new designs, not a commercial vendor.

PR #4787 (2016-07-13): Pr man sens
@kd0aij I added a scale parameters to adjust the throws in full manual for fixed wings.
Do you have any comments on that?

Fixes: #4598
closed 2016-07-13 by AndreasAntener