PX4/PX4-Autopilot #4601: Modernize Docker-based testing infrastructure

LorenzMeier, 2016-05-22:
@AndreasAntener @dagar I would like to modernise the Docker based CI system and consider if moving to a hosted provider (e.g. https://semaphoreci.com) would lower the workload and could allow us to more effectively have people add new tests - or if all we need is a call to bridge some gaps.

Andreas has nicely documented the Docker setup: http://dev.px4.io/advanced-docker.html

I would like to add a couple new tests, but right now I'm missing some steps in between to understand how to get there:

  * Add Snapdragon build test
  * Add SITL run with address sanitizer enabled
  * Expand the SITL reporting further

There is some potential to throw more resources at the problem now that the dev team is growing rapidly. I'm mostly looking for your feedback on best approach, pain points and what you feel you can contribute.

LorenzMeier, 2016-05-22:
@bkueng FYI

dagar, 2016-05-22:
I'll take a look at semaphoreci. I was also thinking about docker containers for the firmware and qgc build environments. Right now travis-ci wastes a lot of time just installing packages every single build.

julianoes, 2016-05-26:
Referencing the discussion concerning adding Snapdragon to the build:
https://github.com/PX4/Firmware/commit/8845070d89644e61de8810f4aad6d17dda5555bc

dagar, 2016-05-30:
I've been exploring how best to handle docker based build environments on both semaphore-ci and travis-ci. First I'm going to move some of the travis-ci builds to docker, then I'll look at parallelizing builds and testing across both travis-ci and semaphore-ci.

dagar, 2016-06-05:
Full `make check` running on semaphore-ci in the px4-dev-base container. So far semaphore-ci seems about twice as fast as travis-ci, but it might be because it's the middle of the night on a weekend...

dagar, 2016-06-14:
I've completely moved the travis-ci linux builds to docker and have basic builds working on semaphore-ci and circle-ci with px4io/px4-dev-base. Next I'm going to get the SITL mission tests running on either semaphore-ci or circle-ci.

The snappy docker image is ~9GB which is just way too big for these services, so I think it'll have to go on jenkins for now.

mcharleb, 2016-06-14:
@dagar I may have pruned down the SDK install (It was installing Android SDK and Eclipse as part of silent install). How small do the images have to be for semaphore-ci and circle-ci?

dagar, 2016-06-14:
Semaphore-ci is ~9.2GB. So that's docker image + full source + build artifacts.
I saw a 500MB directory of logs at /home/docker1000/Qualcomm/Hexagon_SDK/3.0/tools/Installer_logs that I assume can also be deleted.

The other issue is the amount of time each build is going to spend just downloading the image. The basic nuttx/sitl image is ~600MB.

dagar, 2016-06-14:
I've never looked at the raspberry pi stuff. I suppose we should get that going. If it's just the gcc arm hf packages it should be pretty small overall. Maybe ~500MB?

@mhkabir interested in putting together a Dockerfile for raspberry pi builds?
https://github.com/PX4/containers/blob/master/docker/px4-dev/DockerfileBase

mcharleb, 2016-06-15:
@dagar @julianoes I added a -trim option to https://github.com/ATLFlight/cross_toolchain/blob/master/installv3.sh 

You can now do
```
git clone https://github.com/ATLFlight/cross_toolchain
cd cross_toolchain
./installv3.sh -trim ~/myinstallroot
```
This will install the Hexagon SDK and tools without the Android or Eclipse add ons to ~/myinstallroot/Qualcomm/... and then remove unnecessary files from the SDK and Tools to reduce the install size.

mcharleb, 2016-06-16:
@dagar @julianoes The trusty sysroot is now down to 278M. The total should now be Host OS (I'm assuming that is Ubuntu 14.04 and ~500MB) + sysroot (278MB) + Hexagon SDK (101MB) and Hexagon Tools (244MB)

The total Docker image size should now be about 1.2GB

dagar, 2016-06-16:
That's quite an improvement and should be an acceptable size to work on any of the free ci services.

julianoes, 2016-06-20:
Yay thanks @dagar.