PX4/PX4-Autopilot #4604: nuttx_v3:Cmake nuttx patches, configs and src depandancies.

davids5, 2016-05-22:
@dagar FYI: @LorenzMeier 
[From discussion:](https://github.com/PX4/Firmware/commit/e200828a0c491ac64cf8b34ed24f5fa3a597ff55#commitcomment-17572434) 

The issues had to do with the px4pio sub build. I think it is caused by the fact that nuttx_v3 supports the booloaders and has a CONFIG (AKA board) and an nuttx_config (nsh|bootloader) and master does not.

I would really appreciate some help on this from you as it applies to nuttx_v3. The goal would be to achieve the following:

Treat nuttx-config/, nuttx-patches and Firmware/Nuttx as proper dependencies. 
Changes in nuttx-config/, nuttx-patches must cause a clean rebuild. Changed under Firmware/Nuttx (src) would be incremental.

Also adding back the target to run make oldconfig, make menuconfig that got lost in the cmake translation on nuttx_next, would be super helpful.

It copies in the nuttx-config/ runs make oldconfig, make menuconfig then copies the.config back to nuttx-config//defconfig

dagar, 2016-11-08:
We talked about this briefly a few weeks ago and I think the consensus was that pulling nuttx dependencies into PX4 cmake wouldn't work very well. Is there anything else on this you'd still like or need?

davids5, 2016-11-08:
Yes. There is. 

I agree that setting the whole nuttx tree as a dependency, seems like it would bog the build for most devs, when I only need it. - 

But if you can have a look at nuttx_v3 and check the above list of dependencies (TODO) works correctly and then ensure that rsync does a copy on changes would help.

Thank you!

LorenzMeier, 2016-11-19:
1.6.0 is all about NuttX updates.

davids5, 2016-11-19:
@dagar - we need to resolve the || build issues. 

https://travis-ci.org/PX4/Firmware/jobs/177273456

dagar, 2016-11-20:
I'll try to reproduce locally.

LorenzMeier, 2016-12-22:
Finished!

closed 2016-12-22 by LorenzMeier