PX4/PX4-Autopilot #4605: GPIO on Pixhawk

kerail, 2016-05-22:
Hello. I'm trying to read data from AUX GPIO's. 
As i can see they are connected to FMU, so here is my code:
```
_gpio_fd = open(PX4FMU_DEVICE_PATH, 0);
if(_gpio_fd < 0) {
 printf("Error opening FMU\n");
 return;
}
if(ioctl(_gpio_fd, GPIO_SET_INPUT, 1)<0) {
 printf("Error setting as input\n");
 return;
}
uint32_t gpio;
if(ioctl(_gpio_fd, GPIO_GET, (unsigned long) &gpio)) {
 printf("Error reading\n");
 return;
}
printf("GPIO: %d\n", gpio);
```

But i always get `GPIO: 0` as answer, when connected Signal pin to 5V pin of AUX (i tried all 6 to be sure that pin choose is not a mistake);

mhkabir, 2016-05-23:
Set the FMU mode to GPIO. 

```fmu mode_gpio```

The other option would be to disable the FMU driver and directly  access the pins. See the camera trigger driver.

kerail, 2016-05-24:
Seems that fmu mode_gpio changes nothing.

Direct call of stm32_gpio_read and stm32_gpio_write does the trick. 
But when GPIO PULLDOWN input mode is not working for me.

LorenzMeier, 2016-12-28:
Closing as outdated.

closed 2016-12-28 by LorenzMeier