PX4/PX4-Autopilot #4607: Repeated RTL

ender-chen, 2016-05-23:
When Lost RC signal. It will go into RTL. But when it touched land, It will detect takeoff. and go into RTL again.

julianoes, 2016-05-26:
Thanks for reporting. Is this easily reproducible in SITL or how did you find this?

ender-chen, 2016-05-31:
I test it with my plane. I just shutdown my rc controller, but it just repeated rtl. YES, I think you can reproducible in SITL.

julianoes, 2016-05-31:
@ender-chen: can you share logs please and watch out for messages like "Takeoff detected" or "land detected" in QGC?

I would guess that the problem is that landing is not detected (yet) in which case RTL is triggered.

ender-chen, 2016-05-31:
OK, Tomorrow I will test it again. I think I have lost logs.

julianoes, 2016-07-13:
Fixed in #4559.

closed 2016-07-13 by julianoes