PX4/PX4-Autopilot #4609: mavlink stop-all doesn't close all instances when called from SD card

vvranjek, 2016-05-23:
I am running (almost) latest master branch.

I have on my SD
`
echo STOPPING MAVLINK
mavlink stop-all

echo sleeping for 5 secs
usleep 5000000

echo Turning on nshterm on ACM0
nshterm /dev/ttyACM0 &

echo Startin mavLink
mavlink start -b 115200 -d /dev/ttyS1
`

And I am left with 2 instances of mavlink running:
`
228 mavlink_if0                   195  2.948  2308/ 2792 100 (100)  READY 
229 mavlink_rcv_if0                 0  0.000   796/ 2096 175 (175)  w:sem 
170 mavlink_if2                    56  0.000  2284/ 2792 100 (100)  w:sig 
`

mhkabir, 2016-05-23:
Check rcS. Likely mavlink instances are being started after ```extras.txt ``` is run.

vvranjek, 2016-05-23:
Nope, everything stock there, the only mavlink commands after extras.txt is run is 'mavlink boot_complete'.
Also, putting 'mavlink stop-all' one after another fixes the problem. Like this:

echo STOPPING MAVLINK
mavlink stop-all
echo sleeping for 5 secs
usleep 5000000
mavlink stop-all
...
I will try tomorrow with the latest master (if it compiles, becuase today's master had an error and I didn't have time to fix it).

dagar, 2016-11-08:
@vvranjek still an issue? 
mavlink stop-all seems to work for me.

vvranjek, 2016-11-08:
No. I think the issue was that something else was using one of the serial ports, thus mavlink was unable to terminate the instance.

closed 2016-11-08 by dagar