PX4/PX4-Autopilot #4610: Flaps and full-wing flaps on auto takeoff

JakeLeaf, 2016-05-23:
I'd like to take a look at adding flaps to auto takeoff as well as full wing flaps (flaps and ailerons down) for heavy loads. Happy to try getting to work but some direction as to where to look/start would be a big help.
Assigning flaps (or really, full wing flap) amounts to a knob for testing would be how I'd approach it to give some base deflection amounts for auto takeoffs and to come up with a few starting points for different airframes.
 
The spoileron feature on auto land works really well. I can't seem to find it in the code so if someone can point me there that would likely be helpful for the takeoff (mixer is crucial but as set up currently works every time and I haven't changed anything there).

Thanks,

Jake

dagar, 2016-05-23:
Auto takeoff is handled here - https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1605

Take a look at auto land to see how to deploy flaps - https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1363

dagar, 2016-05-23:
Would you split the params and have different flap/flaperon values for takeoff and landing?

JakeLeaf, 2016-05-24:
Thanks for pointing it out. I'm going to start going through it.

I would definitely have separate params. For really heavy takeoffs, fully extended flaps (within their useful range) are great to get off the ground but full wing flaps are the true ideal setup to just make it that much easier to shorten takeoffs distances.

On the same note when landing if we crank the spoilerons too high it's like throwing a parachute out the back to slow down which is good except the nose needs to be managed a bit. A conservative but really reliable setting is just keeping the params set as is even though a lot of optimisation is possible.

Thanks again for pointing out where this is. It is hugely beneficial and not used too frequently to the extent at which it could be.

tubeme, 2016-05-26:
Very good ideas! We are working on autonomous take off and landing on an airstrip with flaps and retracts so giving much functionality for control in this phases of the flight is great.

dagar, 2017-04-12:
Hi @JakeLeaf, did you make any progress with this?

dagar, 2017-04-28:
I can implement this as part of the existing takeoff code if someone is able to test.

see3peoh, 2017-05-09:
This would be awesome for heavy takeoffs. I'm out of town for awhile but would definitely be willing to help. This would be really useful!

closed 2018-02-21 by PX4BuildBot

dagar, 2018-04-13:
> Can't a rudimentary fix be to add _att_sp.apply_flaps = true to auto
> takeoff code to force flaps on during auto takeoff? Would be concerned that
> that flag enables flaperons automatically for auto land so may need to
> separate it somehow since if using flaperaons it will potentially deploy
> airbrakes which is definitely not wanted on takeoff. Maybe invert
> index_airbrake on takeoff somehow?

Yes that's really all it would take to get started. As you said the main
issue is that apply_flags currently deploys both flaps and flaperons
(airbrakes in the mixer) and the same scaling may not be appropriate for
both landing and takeoff(see
https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/FixedwingAttitudeControl.cpp#L870
).

Giving the position controller full access (not just a boolean) to both
independently might enable more interesting things down the line. The
position controller knows what you're trying to do (takeoff, land, fly
normally), but also has a full understanding of airspeed and altitude
management. I'd like to experiment with giving the position controller
access to deploy both flaps and airbrakes automatically and incrementally.

ryanjAA, 2018-04-13:
If you build i will test fly it :) 

All seriousness though, totally right with you on pos controller and some logic as to what to actually do given the provided information at the time. Either way - will get some foam planes up in the sky to real world test every which way.