PX4/PX4-Autopilot #4612: VTOL decreases altitude during transition to fw

nkhoit, 2016-05-23:
http://logs.uaventure.com/view/aNd5eCeXsfm4Bfn8WLh2ML#Local_Z_PLOT

A look at LPSP.VZ shows the climb rate controller commanded a negative climb rate of 2 m/s which is equal to our MPC_Z_VEL_MAX. 

[image]

tubeme, 2016-05-26:
We've had this phenomenon in Altitude mode transition two weeks ago. But no problems in Stabilized at the same time. In the coming days, when we have a weather window, we will fly test it again and will report with LOG graphics as well.

sanderux, 2016-07-07:
@nkhoit what kind of vehicle are you flying?

sanderux, 2016-07-07:
Also, i see you have added a custom param VT_MC_TO_FW_PIT that is set to -6
And you have FW_PSP_OFF set to -6 too, in most cases you would want that positive, realy depends on your airframe

closed 2016-08-31 by sanderux