PX4/PX4-Autopilot #4614: Flaperons change neutral aileron PWM value

JakeLeaf, 2016-05-24:
When enabling flaperons (spoilerons) and assigning them to aux 2 it works perfectly for auto landing. It's actually spot on. The problem is that it changes the centre pwm value from 1500 to 1420 for the ailerons. Tested multiple times, multiple type of servos and multiple versions of the firmware. If you mechanically centre your ailerons at 1500 pwm during setup it become an issue because after the fact you need to reset them to be centered at 1420 and adjusting this by trim shouldn't be the fix. It's easy to recreate, just load up a plane with flaperon support (flaps, ailerons in the mixer) and don't assign the flaps or aux2 to a channel.. Ailerons will be centred at 1500. Open QGC and set aux2 and flaps to channel 9 and it will instantly change down to 1420. It's an easy mechanical fix but beyond being a bit of an annoyance, I can't seem to understand why it is occurring.

edsellar, 2016-05-24:
@JakeLeaf Confirmed it is not working correctly - does the same thing to me as well. Just tested now.

dagar, 2016-05-25:
I'll try to replicate this on my bixler 3 this weekend and take a look at what's happening.

kd0aij, 2016-06-01:
@dagar This may be the cause if the flaperon setup changes pwm_min/max:
https://github.com/PX4/Firmware/issues/4650#issuecomment-222495368

kd0aij, 2016-10-16:
I can't reproduce this with my flaperon setup. Using QGC to assign the aux2 and flaps channels has no effect on the aileron neutral position in manual or stabilize modes.

kd0aij, 2016-10-17:
It's not clear whether the flaperons function is implemented correctly in the FW controllers, so I mapped the FLAPS channel instead of aux2 and changed my mixer to use the FLAPS actuator channel instead of the AIRBRAKE channel.

dagar, 2016-10-17:
@JakeLeaf can you test this again with current master?
This particular issue may have been addressed by a flap/flaperon cleanup last month.

dagar, 2016-11-08:
I believe this has been resolved. Please reopen if it's still an issue.

closed 2016-11-08 by dagar