PX4/PX4-Autopilot #4616: Airspeed sensor calibration useability

sanderux, 2016-05-24:
Airspeed sensors remain quite inacurate when facing different weather conditions. For best results an airspeed sensor should be calibrated before every flight. For end users this can be a hard task that is easilly forgotten.

I would like to some form of auto calibration. Perhaps by using a second airspeed sensor dedicated to measuring only ambient pressure (mounted so it will never read wind)

Any thoughts welcome.

sanderux, 2016-05-24:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Airspeed/Airspeed_Calibration.cpp

As per @tumbili's suggestion

dagar, 2016-05-24:
This in flight autocal is interesting. Does anyone here have first hand experience with it?

Personally I'd like to take a step back, gather data, and take a closer look at what's actually happening. Something like flying a large loiter over the full range of min, cruise, max airspeeds (several cycles at each) and then comparing to ground speed.

tumbili, 2016-05-24:
@dagar I think @priseborough is the best person to ask because he's the one who has written the code. If this is something that works well then I'm happy to also implement it for PX4.

priseborough, 2016-05-25:
There are two parts to calibration - offset and scale factor. Offset should be calibrated on the ground using a pitot/static cover that ensures the pitot and static ports see the same pressure. Scale factor can be calibrated in-flight using a simple EKF like the one I developed for ArduPilot. 

The scale factor calibration requires an estimate of the of air density or the ratio between EAS and TAS.

The biggest contributor to EAS measurement scale factor error is aerodynamic. If the static ports are in a region of accelerated flow (eg above a wing or canopy), the airspeed will read high, if they are in a region of decelerated flow (in front of the nose) then it will read low. This is normally referred to as 'position error'

tubeme, 2016-05-26:
@sanderux My opinion is that we have to dig deeper in to the problem. And I must say it is not a software one, but rather hardware. We reported last year, a very nasty misbehavior of the airspeed sensor and had a lot of crashes because of this. After we understood that the Sensor that everybody that made HW was using should not be placed close to the battery and close to any emitting electronics our problems got cured. We also understood the I2C is very conductive you should be careful where you place your i2c spliter and where are the wires passing around the craft. In one case we've had a 3m wingspan plane so for the i2c we used special grade thin silicon FTP cable and worked well.

Also there was a problem discovered by mr. Arsov with the 5v vs 3v sensors, the current 5V implementation of the airspeed sensor and the problems mismatching voltages could generate with the current i2c splutter. This again could be source of problems if not handled properly. 

BTW after we take in consideration all of this installation conditions and we are always careful about how and where we place the sensor the spliter and the wires we've had a very consistent results with the airpseed sensor in all weather conditions without the need of calibration. We are trying Eagle Tree airspeed sensor V3 now and my first impresion is that it gives better results. May be less sensitive a bit, probably because it is analog but this gives kind of smoother graphics.

We will have further flight testing on the ET sensor and will report back.

jcooper21, 2016-05-27:
The auto calibration APM does causes more issues because if you don't cover the pitot tube on boot perfectly then your calibration is out and you have wind issues. I have had serious issues on take off and crashes because of that.

sanderux, 2016-05-27:
Could the cover be detected?

sanderux, 2016-05-27:
@tubeme i am also tesing the eagle tree v3 and i did take car to isolate the circuit from other hardware. The results are quiter consistent but not very acurate. I have even seen 15ms readings that (based on ground speed flying a circle) must have been closer to 18ms.

For my endurance setups i need not only stable readings but also acurate readings.

priseborough, 2016-05-27:
You are taking about the offset calibration, not the in-flight scale factor calibration. Covering the probe on the ground is an operator responsibility.

jcooper21, 2016-05-27:
@priseborough how does sensefly and Trimble implement this. Both of them seem to work very well from my testing and neither have to cover the pitot tube. I know sensefly does its calibration as soon as it hits take off alitude as described below. 
[image]

proficnc, 2016-05-27:
That would be a PITA.... 
I would rather calibrate by covering...

priseborough, 2016-05-27:
@jcooper21 Not a good idea, the autopilot gains could be too high or the plane could stall before the calibration completed given the errors with our sensor types. Also their method will be bad at separating offset from scale factor. Pitot covers are cheap and easy to use. They should be part of the standard pre-flight checklist for a fixed-wing UAV operator and be attached to a large hi-vis ribbon so they are not missed.

sanderux, 2016-05-27:
From an industry perspective, using a cover would always have to be based on a sensor that can tell if the cover is attached. 

Considering the calibration i think flying a circle is a good idea, this will let you calibrate scale the airspeed based on ground speed as flying an exact circle should void the wind factor.

sanderux, 2016-05-27:
Concerning the crash risk we could set a minimum throttle level to eliminate this.

priseborough, 2016-05-27:
Flying in a circle is something that is used in APM to calibrate scale factor error and it works well and in fact when using the EKF it does not have to be a circle. The place to check (and calibrate if necessary) offset is on the ground before you take off.

sanderux, 2016-05-27:
It seems to me that, checking on the ground would calibrate the static with the dynamic pressure sensor. calibrating in the air (scaling) would give you an exact reading based on ground speed. I'm trying to achieve 2 things; 
1. make the sensor more reliable and consistent. 
2. make the reading more accurate

it seems to me that performing both types of calibration would be required to achieve both. Or do you think we could achieve this by only performing a ground based calibration?

tubeme, 2016-05-27:
@sanderux Sorry I did not clarify. I meant Consistent and Accurate readings. When on the ground in calm weather we get this ussual 0-1m/s readings (we have more than 10 sensors for testing). Your jumps to 15m/s are not a calibration problem! When we have calibration problem then we have consistently offset readings like 10m/s and it stays at this on the ground. You can try calibrating the Pix Airpseed sensor and then connect the Eagle Tree V3 sensor. You will see a reading at about 50m/s because of the different scaling of the sensor. 

We've had a lot of situations like yours with spikes to 15-20m/s while staying on the ground. And in all cases these were HW related. When the sensor is close to the battery we've had spikes. We've had cases of airspeed sensor chip not very well soldered to the board, again spikes. We've had i2c wire or rather connector problems... spikes. We've had the i2c split er close to emitting electronics... spikes. 

@LorenzMeier @sanderux As we discussed with Lorenz long time ago, airspeed reading is one of key components in an autopilot system and I agreed with him. The calibration process of ANY sensor and any SYSTEM should be done in controlled environment! Otherwise it is not an instrument calibration but rather a snapshot of a dynamic system condition. If an instrument is well designed it needs calibration but once in a wile even it could sustain a longer periods of time without the need of calibration. The flight is a very dynamic condition so I don't think it is good idea to mess with a such important key element in the air.

If we want to really see in and check in flight if the airspeed sensor is OK there is a way  that could be done. If the plane makes one or too circles PX4 could detect wind direction by the readings of the airspeed sensor and match this to the corresponding GPS ground speed readings. If the math did not match then we have a problem.

Yesterday I got the chance of getting inside the Eurocopter Panther and there was a flight mode for rescue, with the same procedure for wind detection. Once over the target, you get to the hovering altitude that you want, you push a button and then the helicopter makes a circle, detects the wind then directs the heli against the wind and in front of the target that you could see from the lower window at your feet and it hovers there in PosCtl.

sanderux, 2016-05-27:
@tubeme i'm not having 15m/s spikes anymore. that was on the 3dr sensor before i added a level shifter, i'm getting the same readings you are. after calibration it jumps between 0-2ms and uncalibrated in other conditions it's offset to about 5ms. this means the reading might be consistent throughout the flight, but it is inaccurate.

tubeme, 2016-05-27:
@sanderux Hmmm we've had no such case for a very long time.. Did you notice what is the condition that creates this offset? When we fly in strong weather, like 6-7 m/s then on the ground we get jumps between 0-5 m/s probably because the wind is not so laminar flow of air against the pitot tube.

priseborough, 2016-05-27:
5 m/s at zero airspeed equates to 15Pa of pressure offset which is well within the spec for sensor drift with temperature. How important this is depends on the speed of your flight vehicle. At 15 m/s, the dynamic pressure is 138 Pa, so an additional 15Pa of measured pressure would equate to an airspeed error of +0.8 m/s. At higher speeds it will be less. Also depending on how the filtering and airspeed calculation is done, it may be rectifying noise which can cause the reading to be higher on the ground.

tubeme, 2016-05-27:
I agree with @priseborough that we don't need such accuracy probably. In terms of mission and real performance we measure the distances by ground speed ie. the real kilometers flown. And the GPS reading is pretty accurate, getting even better with new Ublox modules and better antennas. We need the airspeed reading for sustaining flight dynamic so just give 1-2m/s reserve speed in the min_airspeed and you are well on the safe side. With this strategy we have good overall results.

sanderux, 2016-05-27:
@priseborough have a look at this log: http://logs.uaventure.com/view/jg2rHZDRmAboYLiqyFVuge
i used the KML file to calculate ground speed and came to around 18m/s.

Also, the accelerometer seems to think the value was closer to that. this misreading meant the difference between a 30amp draw and a 10amp draw.

jcooper21, 2016-05-27:
@tubeme your circling concept sounds the best to me. As a heavy user of both PX4 and APM I can say overall I have less airspeed issues with PX4.

AndreasAntener, 2016-10-04:
@priseborough seems like we're moving things again on this topic :) You were talking about the "position error" being the major driver behind the scale factor. But assuming we don't have a position error OR it is known for a certain airframe. What is still left that influences the scaling? I'm currently gathering data to get an idea about the temperature dependency.

priseborough, 2016-10-04:
Offset drift with temperature is a bigger problem than scale factor for the low speeds that most small UAV's fly. It's easy to check on the ground.

AndreasAntener, 2016-10-04:
Why checking? We need to understand the dependency so we can compensate. The temperature from boot to in air might change 30 degrees.

AndreasAntener, 2016-10-04:
Assuming this is possible somehow.. How big can the offset drift with temperature be for the MS4525DO?

Looking at possible outcomes with a 30pa error:

actual airspeed | measured | error
----- | ----- | -----
0 | 7 | 7
7 | 9.9 | 2.9
9.9 | 12.1 | 2.2
15.1 | 16.6 | 1.5

with a negative error, -30pa:

actual airspeed | measured | error
----- | ----- | -----
0 | -7 | 7
7 | 0 | 7
9.9 | 7 | 2.9
15.1 | 13.4 | 1.7

maksimpiriyev, 2017-01-18:
I just found this quite interesting discussion,I propose an algorithm,would be glad if @priseborough  could also comment, to make 2 loiters with 2 different airspeeds as v1,v2. In the first loiter auto calibration adjusts the "offset",in the second with different airspeed v2 , it adjusts "scale". Example the drone loiters with airspeed v1=10m/s ,adjusts "offset", then makes another loiter with v2=15m/s.
Note that since changing scale in the 2nd loiter will probably effect the previous "offset" adjusted result,there should be a 3rd step(without loitering) to readjust "offset","scale" at the same time until they converge.
This still doesnt elminate the risk as @priseborough  mentioned "Not a good idea, the autopilot gains could be too high or the plane could stall before the calibration completed given the errors with our sensor types" meanwhile might reduce its probability and improve the precision in some applications.
I plan to implement and test it but i have few questions i believe @dagar could answer them.
1.what would be the right mission frame-command combination , as px4 doesnt implement the loiter with turns, should it be implemented first ?
2.since wind is also estimable,would it effect the flight performance if the wind is estimated with the current flight controller.
3.another thing here https://github.com/PX4/Firmware/blob/b6b889260689cb340d6815fc68aea831751f48bc/src/drivers/airspeed/airspeed.cpp#L277  set/get "scale" is ignored

priseborough, 2017-01-18:
Trying to estimate both offset and scale factor in-flight is an unnecessary complication.

Sensor zero offset error can be calibrated on the ground with the pitot tube covered.

Scale factor error, which is mostly caused by positional error and not having static pressure flow at the static ports can be calibrated in the air using a simple 3-state EKF like the one I implemented on ArduPilot over two years ago: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Airspeed/Airspeed_Calibration.cpp

dagar, 2017-01-18:
@priseborough naive question, but could we (should we?) do something similar in EKF2 or would it be better standalone?

priseborough, 2017-01-19:
I would rather have it implemented in the library where the airspeed is calculated from the pressure as it would require adding a state to the EKF that has nothing to do with navigation.

tumbili, 2017-01-19:
@dagar I already wrote a wind and airspeed scale estimator.
https://github.com/tumbili/Firmware/tree/wind_estimator

I gave it a try in SITL and the results seemed good. There is also a derivation script in python here
https://github.com/tumbili/Firmware/blob/wind_estimator/src/lib/wind_estimator/python/wind_est_derivation.py

Would be nice if somebody from the PX4 team could check the derivation.
I already integrated it into ekf2 but it would require a rebase which shouldn't be painful.

tumbili, 2017-01-19:
@dagar FYI https://github.com/PX4/Firmware/pull/6386

dagar, 2017-01-19:
Thanks Roman! What other surprises do we have hidden in old branches?

tubeme, 2017-01-19:
@dagar Here is a question. During the calibration in 0 m/s speed the PX4 reading from the static sensor equals the reading from the pitot which equals some atmosphere pressure. Or they are two different sensors for the static and pitot and have their "0" calibrated? If there is wind present, the pitot sensor is bouncing so we don't know where the 0 is right?

If this is the case then there is approach to the problem for autocalibration on the ground.

The bouncing up and down have some amplitude dependent on the wind. So if we keep track of this amplitudes for a while - 30 - 50 seconds then we will have a statisticaly correct min value ie the "0" and statisticaly correct max value ie the wind bursts speed. So is it possible that we can calibrate in windy weather with such a method?

The static pressure reading should be the same even in wind, correct?

Vasil

priseborough, 2017-01-20:
The airspeed sensor is differential. It measures the gauge pressure between the side of the sensor connected to the static ports (small holes around the circumference of the tube) and the side of the sensor connected to the total pressure port (the small hole facing forward).

tumbili, 2017-01-20:
@dagar Hehe, I think I had a branch in which I implemented the perfect autopilot but I can't find it anymore ;-)

maksimpiriyev, 2017-01-31:
I tried pitot cover with modifying the airspeed calibration code.After the calibration it started to bounce between 0-2m/s with freq=0.2-1 then after some minutes offset slightly moved up with airspeed bouncing between 1-3m/s. As @AndreasAntener  mentioned this probably would not cause a problem in high speeds. Another idea might be useful to model the natural frequency (or the damping constant) of the sensor, to detect existence of the non-wind in realtime.

closed 2017-09-06 by dagar