PX4/PX4-Autopilot #4619: Add WIFI TX on/off control for Pixracer

kd0aij, 2016-05-24:
Proposal: add new parameter with options:
1) OFF
2) ON
3) RC control

The RC control option will enable/disable WIFI based on the state of the channel specified by another new params:
RC_MAP_WIFI
RC_WIFI_THR

LorenzMeier, 2016-05-24:
Makes sense - would you like to implement this in the FMU app using the power control GPIO for the Wifi? I would propose a 4th state: OFF ON ARM. This should first send a string that its shutting off Wifi, wait for 200 ms, then disable wifi once you arm the system.

I honestly believe that is going to be the optimal trade-off for a lot of users.

kd0aij, 2016-05-24:
great point. I think I'll implement states 0-3 in the new list (above) first, and leave RC control for later, it it seems necessary.
And yes, I planned to use the TX on/off GPIO to implement it.

kd0aij, 2016-05-24:
I had RC control in there to check whether RSSI decreases with WIFI on. But I can just do 2 controlled test flights to obtain the same info using FrSky logging at the TX.

kd0aij, 2016-05-24:
BTW, I'm doing this at Phil's suggestion, and I see "off while armed" was what he asked for in the first place. Sorry Phil.

kd0aij, 2016-05-24:
Another note: during a mission test on the weekend, I gave my spotter the Nexus 7 to follow the progress and the display came in very handy when the quad wound up shooting into the sky, then blowing too far away after I switched back to manual. When I found that I was unable to get oriented by rolling and pitching the quad, I asked my spotter to tell me what heading was displayed on the GCS, and knowing that I was immediately reoriented and able to fly back. (Didn't want to trust RTL at that point)

kd0aij, 2016-05-24:
https://github.com/PX4/Firmware/pull/4622

closed 2016-08-07 by kd0aij