PX4/PX4-Autopilot #4634: new logger performance on nuttx_v3: Sandisk 8GB Class 4 SD card, PixRacer

kd0aij, 2016-05-25:
It looks like the new logger is working very well in nuttx_v3. Even though I had to reduce buffer size to 12K due to less RAM available at startup, I'm not seeing any dropouts at >100kB/sec. 

It seems we need more than 10K bytes free to be able to run commands like top and "commander arm/disarm". Otherwise the console responds "command not found". 

@LorenzMeier @davids5 Is there a way for the logger to determine at startup how much memory it can (safely) allocate for a buffer? I'm sure we need some minimum amount of heap available for running commands, but don't know if a static analysis can determine how much.

@bkueng Is my highwater mark code still in there, and if so, how can I display the results?

```
nsh> free
             total       used       free    largest
Mem:        227008     214496      12512       9376
```
"top" when logging at ~105kB/sec:
```
 185 logger                       2246  4.267  2856/ 3140 250 (250)  w:sig 
 188 log_writer                    256  1.839   528/ 1020  60 ( 60)  w:sem 

nsh> commander disarm
nsh> INFO  closed logfile: /fs/microsd/log/sess001/log001.ulg, bytes written: 1845155
commander arm
nsh> INFO  start log
INFO  Opened log file: /fs/microsd/log/sess001/log002.ulg
commander disarm
nsh> INFO  closed logfile: /fs/microsd/log/sess001/log002.ulg, bytes written: 16310084
```
after logging for about 2 minutes:
```
sd fsync: 43 events, 0 overruns, 1652416us elapsed, 38428us avg, min 7959us max 127697us 47548.926us s
sd write: 4422 events, 0 overruns, 70144198us elapsed, 15862us avg, min 10us max 380912us 14752.410uss
```
sd_bench:
```
nsh> sd_bench
INFO  Using block size = 4096 bytes, sync=0
INFO  
INFO  Testing Sequential Write Speed...
INFO    Run  0:   215.19 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 8 ms
INFO    Run  1:   213.40 KB/s, max write time: 47 ms (=  85.11 KB/s), fsync: 8 ms
INFO    Run  2:   213.79 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 8 ms
INFO    Run  3:   215.91 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 8 ms
INFO    Run  4:   210.72 KB/s, max write time: 40 ms (= 100.00 KB/s), fsync: 8 ms
INFO    Avg   :   213.80 KB/s
nsh> sd_bench -s
INFO  Using block size = 4096 bytes, sync=1
INFO  
INFO  Testing Sequential Write Speed...
INFO    Run  0:   183.13 KB/s, max write time: 48 ms (=  83.33 KB/s), fsync: 429 ms
INFO    Run  1:   184.48 KB/s, max write time: 33 ms (= 121.21 KB/s), fsync: 410 ms
INFO    Run  2:   187.36 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 403 ms
INFO    Run  3:   184.84 KB/s, max write time: 35 ms (= 114.29 KB/s), fsync: 429 ms
INFO    Run  4:   184.36 KB/s, max write time: 36 ms (= 111.11 KB/s), fsync: 442 ms
INFO    Avg   :   184.83 KB/s
```

bkueng, 2016-05-25:
looks great! Did you let sd_bench run for longer (-r 50)? For the logger you can do `logger status` or uncomment
```
//#define DBGPRINT //write status output every few seconds
```
in `logger.cpp` to see highwater output.

davids5, 2016-05-25:
@kd0aij 

We probably need a stack size tuning session - to validate the stack sizes and ensure they are not taking up too much RAM from startup.

see the free_main aa way to getting the free RAM.

kd0aij, 2016-05-25:
@bkueng Thanks, I should have remembered that. It looks like 12K is pretty good for 100K bytes/sec, but the highwater mark got above 11000 in this test:
```
nsh> sd_bench -r 50
INFO  Using block size = 4096 bytes, sync=0
INFO  
INFO  Testing Sequential Write Speed...
INFO    Run  0:   211.50 KB/s, max write time: 35 ms (= 114.29 KB/s), fsync: 18 ms
INFO    Run  1:   209.82 KB/s, max write time: 76 ms (=  52.63 KB/s), fsync: 8 ms
INFO    Run  2:   212.79 KB/s, max write time: 39 ms (= 102.56 KB/s), fsync: 8 ms
INFO    Run  3:   212.77 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 8 ms
INFO    Run  4:   210.76 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 9 ms
INFO    Run  5:   212.77 KB/s, max write time: 36 ms (= 111.11 KB/s), fsync: 8 ms
INFO    Run  6:   210.96 KB/s, max write time: 55 ms (=  72.73 KB/s), fsync: 8 ms
INFO    Run  7:   213.92 KB/s, max write time: 40 ms (= 100.00 KB/s), fsync: 9 ms
INFO    Run  8:   213.08 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 8 ms
INFO    Run  9:   214.97 KB/s, max write time: 41 ms (=  97.56 KB/s), fsync: 8 ms
INFO    Run 10:   211.00 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 8 ms
INFO    Run 11:   211.20 KB/s, max write time: 42 ms (=  95.24 KB/s), fsync: 18 ms
INFO    Run 12:   209.93 KB/s, max write time: 55 ms (=  72.73 KB/s), fsync: 9 ms
INFO    Run 13:   211.49 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 8 ms
INFO    Run 14:   210.57 KB/s, max write time: 47 ms (=  85.11 KB/s), fsync: 8 ms
INFO    Run 15:   211.63 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 9 ms
INFO    Run 16:   211.55 KB/s, max write time: 40 ms (= 100.00 KB/s), fsync: 33 ms
INFO    Run 17:   212.98 KB/s, max write time: 41 ms (=  97.56 KB/s), fsync: 8 ms
INFO    Run 18:   210.79 KB/s, max write time: 55 ms (=  72.73 KB/s), fsync: 18 ms
INFO    Run 19:   211.49 KB/s, max write time: 39 ms (= 102.56 KB/s), fsync: 8 ms
INFO    Run 20:   211.53 KB/s, max write time: 50 ms (=  80.00 KB/s), fsync: 8 ms
INFO    Run 21:   210.68 KB/s, max write time: 45 ms (=  88.89 KB/s), fsync: 8 ms
INFO    Run 22:   211.58 KB/s, max write time: 41 ms (=  97.56 KB/s), fsync: 15 ms
INFO    Run 23:   207.55 KB/s, max write time: 47 ms (=  85.11 KB/s), fsync: 8 ms
INFO    Run 24:   213.73 KB/s, max write time: 41 ms (=  97.56 KB/s), fsync: 8 ms
INFO    Run 25:   211.44 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 15 ms
INFO    Run 26:   212.91 KB/s, max write time: 40 ms (= 100.00 KB/s), fsync: 7 ms
INFO    Run 27:   214.39 KB/s, max write time: 41 ms (=  97.56 KB/s), fsync: 8 ms
INFO    Run 28:   212.17 KB/s, max write time: 42 ms (=  95.24 KB/s), fsync: 8 ms
INFO    Run 29:   209.89 KB/s, max write time: 47 ms (=  85.11 KB/s), fsync: 18 ms
INFO    Run 30:   212.99 KB/s, max write time: 42 ms (=  95.24 KB/s), fsync: 8 ms
INFO    Run 31:   215.21 KB/s, max write time: 42 ms (=  95.24 KB/s), fsync: 18 ms
INFO    Run 32:   213.75 KB/s, max write time: 39 ms (= 102.56 KB/s), fsync: 16 ms
INFO    Run 33:   211.22 KB/s, max write time: 36 ms (= 111.11 KB/s), fsync: 17 ms
INFO    Run 34:   213.57 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 7 ms
INFO    Run 35:   209.01 KB/s, max write time: 55 ms (=  72.73 KB/s), fsync: 8 ms
INFO    Run 36:   213.06 KB/s, max write time: 39 ms (= 102.56 KB/s), fsync: 8 ms
INFO    Run 37:   212.95 KB/s, max write time: 42 ms (=  95.24 KB/s), fsync: 8 ms
INFO    Run 38:   214.36 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 8 ms
INFO    Run 39:   211.40 KB/s, max write time: 50 ms (=  80.00 KB/s), fsync: 15 ms
INFO    Run 40:   210.81 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 9 ms
INFO    Run 41:   211.62 KB/s, max write time: 48 ms (=  83.33 KB/s), fsync: 15 ms
INFO    Run 42:   211.47 KB/s, max write time: 39 ms (= 102.56 KB/s), fsync: 7 ms
INFO    Run 43:   212.33 KB/s, max write time: 36 ms (= 111.11 KB/s), fsync: 9 ms
INFO    Run 44:   210.68 KB/s, max write time: 42 ms (=  95.24 KB/s), fsync: 9 ms
INFO    Run 45:   213.94 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 7 ms
INFO    Run 46:   208.47 KB/s, max write time: 52 ms (=  76.92 KB/s), fsync: 8 ms
INFO    Run 47:   214.64 KB/s, max write time: 42 ms (=  95.24 KB/s), fsync: 8 ms
INFO    Run 48:   211.36 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 8 ms
INFO    Run 49:   213.25 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 8 ms
INFO    Avg   :   211.96 KB/s
nsh> commander arm
nsh> INFO  start log
INFO  Opened log file: /fs/microsd/log/sess001/log001.ulg
logger status
INFO  Running
INFO  Wrote 0.49 MiB (avg 106.03 KiB/s)
INFO  Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 10555 / 12288 B
nsh> logger status
INFO  Running
INFO  Wrote 2.11 MiB (avg 101.68 KiB/s)
INFO  Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 8212 / 12288 B
nsh> logger status
INFO  Running
INFO  Wrote 4.29 MiB (avg 100.96 KiB/s)
INFO  Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 9350 / 12288 B
nsh> logger status
INFO  Running
INFO  Wrote 12.93 MiB (avg 100.60 KiB/s)
INFO  Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 9604 / 12288 B
nsh> logger status
INFO  Running
INFO  Wrote 61.49 MiB (avg 100.45 KiB/s)
INFO  Since last status: dropouts: 0 (max len: 0.000 s), max used buffer: 11232 / 12288 B
nsh> commander disarm
nsh> INFO  closed logfile: /fs/microsd/log/sess001/log001.ulg, bytes written: 67459853
logger status
INFO  Running, but not logging
```

pkocmoud, 2016-05-28:
**Sandisk Extreme PLUS 32GB Class 10 SD card, PixRacer**

nsh> sd_bench -r 50
INFO  Using block size = 4096 bytes, sync=0
INFO
INFO  Testing Sequential Write Speed...
INFO    Run  0:   321.58 KB/s, max write time: 123 ms (=  32.52 KB/s), fsync: 5 ms
INFO    Run  1:   341.67 KB/s, max write time: 118 ms (=  33.90 KB/s), fsync: 5 ms
INFO    Run  2:   335.50 KB/s, max write time: 116 ms (=  34.48 KB/s), fsync: 5 ms
INFO    Run  3:   319.00 KB/s, max write time: 127 ms (=  31.50 KB/s), fsync: 4 ms
INFO    Run  4:   352.62 KB/s, max write time: 116 ms (=  34.48 KB/s), fsync: 5 ms
INFO    Run  5:   328.73 KB/s, max write time: 130 ms (=  30.77 KB/s), fsync: 5 ms
INFO    Run  6:   319.14 KB/s, max write time: 118 ms (=  33.90 KB/s), fsync: 4 ms
INFO    Run  7:   354.94 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 5 ms
INFO    Run  8:   327.84 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 4 ms
INFO    Run  9:   327.16 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 4 ms
INFO    Run 10:   353.72 KB/s, max write time: 115 ms (=  34.78 KB/s), fsync: 5 ms
INFO    Run 11:   322.06 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 5 ms
INFO    Run 12:   327.34 KB/s, max write time: 145 ms (=  27.59 KB/s), fsync: 4 ms
INFO    Run 13:   346.48 KB/s, max write time: 118 ms (=  33.90 KB/s), fsync: 5 ms
INFO    Run 14:   320.04 KB/s, max write time: 130 ms (=  30.77 KB/s), fsync: 5 ms
INFO    Run 15:   349.11 KB/s, max write time: 130 ms (=  30.77 KB/s), fsync: 4 ms
INFO    Run 16:   319.52 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 5 ms
INFO    Run 17:   327.96 KB/s, max write time: 116 ms (=  34.48 KB/s), fsync: 5 ms
INFO    Run 18:   341.70 KB/s, max write time: 131 ms (=  30.53 KB/s), fsync: 5 ms
INFO    Run 19:   310.90 KB/s, max write time: 132 ms (=  30.30 KB/s), fsync: 5 ms
INFO    Run 20:   350.99 KB/s, max write time: 116 ms (=  34.48 KB/s), fsync: 5 ms
INFO    Run 21:   320.14 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 5 ms
INFO    Run 22:   317.67 KB/s, max write time: 131 ms (=  30.53 KB/s), fsync: 5 ms
INFO    Run 23:   356.19 KB/s, max write time: 116 ms (=  34.48 KB/s), fsync: 5 ms
INFO    Run 24:   319.49 KB/s, max write time: 121 ms (=  33.06 KB/s), fsync: 5 ms
INFO    Run 25:   325.92 KB/s, max write time: 124 ms (=  32.26 KB/s), fsync: 5 ms
INFO    Run 26:   339.21 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 9 ms
INFO    Run 27:   313.66 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 5 ms
INFO    Run 28:   353.58 KB/s, max write time: 116 ms (=  34.48 KB/s), fsync: 4 ms
INFO    Run 29:   333.82 KB/s, max write time: 130 ms (=  30.77 KB/s), fsync: 4 ms
INFO    Run 30:   328.28 KB/s, max write time: 115 ms (=  34.78 KB/s), fsync: 4 ms
INFO    Run 31:   344.86 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 4 ms
INFO    Run 32:   326.33 KB/s, max write time: 161 ms (=  24.84 KB/s), fsync: 4 ms
INFO    Run 33:   323.97 KB/s, max write time: 125 ms (=  32.00 KB/s), fsync: 4 ms
INFO    Run 34:   358.08 KB/s, max write time: 116 ms (=  34.48 KB/s), fsync: 4 ms
INFO    Run 35:   334.56 KB/s, max write time: 116 ms (=  34.48 KB/s), fsync: 4 ms
INFO    Run 36:   316.08 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 4 ms
INFO    Run 37:   349.94 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 5 ms
INFO    Run 38:   324.24 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 4 ms
INFO    Run 39:   330.54 KB/s, max write time: 132 ms (=  30.30 KB/s), fsync: 4 ms
INFO    Run 40:   320.18 KB/s, max write time: 145 ms (=  27.59 KB/s), fsync: 4 ms
INFO    Run 41:   324.74 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 3 ms
INFO    Run 42:   341.04 KB/s, max write time: 117 ms (=  34.19 KB/s), fsync: 4 ms
INFO    Run 43:   331.10 KB/s, max write time: 125 ms (=  32.00 KB/s), fsync: 4 ms
INFO    Run 44:   330.83 KB/s, max write time: 118 ms (=  33.90 KB/s), fsync: 4 ms
INFO    Run 45:   307.15 KB/s, max write time: 132 ms (=  30.30 KB/s), fsync: 4 ms
INFO    Run 46:   334.37 KB/s, max write time: 116 ms (=  34.48 KB/s), fsync: 4 ms
INFO    Run 47:   338.84 KB/s, max write time: 144 ms (=  27.78 KB/s), fsync: 4 ms
INFO    Run 48:   312.17 KB/s, max write time: 118 ms (=  33.90 KB/s), fsync: 5 ms
INFO    Run 49:   351.72 KB/s, max write time: 125 ms (=  32.00 KB/s), fsync: 5 ms
INFO    Avg   :   332.09 KB/s

pkocmoud, 2016-05-28:
**Sandisk Pixtor  16GB Class 10 SD card, PixRacer**

sd_bench -r 50
INFO  Using block size = 4096 bytes, sync=0
INFO
INFO  Testing Sequential Write Speed...
INFO    Run  0:   356.67 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 5 ms
INFO    Run  1:   360.41 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run  2:   360.57 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run  3:   363.63 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run  4:   363.03 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 5 ms
INFO    Run  5:   359.87 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 5 ms
INFO    Run  6:   360.70 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run  7:   362.54 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run  8:   360.70 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run  9:   360.96 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 6 ms
INFO    Run 10:   366.41 KB/s, max write time: 20 ms (=200.0000 KB/s), fsync: 5 ms
INFO    Run 11:   359.88 KB/s, max write time: 26 ms (= 153.85 KB/s), fsync: 5 ms
INFO    Run 12:   363.48 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run 13:   359.37 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run 14:   269.71 KB/s, max write time: 193 ms (=  20.73 KB/s), fsync: 31 ms
INFO    Run 15:   256.26 KB/s, max write time: 299 ms (=  13.38 KB/s), fsync: 31 ms
INFO    Run 16:   277.43 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 7 ms
INFO    Run 17:   304.16 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 6 ms
INFO    Run 18:   300.51 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 6 ms
INFO    Run 19:   300.08 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 6 ms
INFO    Run 20:   298.50 KB/s, max write time: 26 ms (= 153.85 KB/s), fsync: 7 ms
INFO    Run 21:   272.58 KB/s, max write time: 219 ms (=  18.26 KB/s), fsync: 7 ms
INFO    Run 22:   268.41 KB/s, max write time: 139 ms (=  28.78 KB/s), fsync: 31 ms
INFO    Run 23:   298.77 KB/s, max write time: 35 ms (= 114.29 KB/s), fsync: 7 ms
INFO    Run 24:   351.60 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 5 ms
INFO    Run 25:   348.20 KB/s, max write time: 90 ms (=  44.44 KB/s), fsync: 5 ms
INFO    Run 26:   336.71 KB/s, max write time: 161 ms (=  24.84 KB/s), fsync: 5 ms
INFO    Run 27:   362.63 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run 28:   362.10 KB/s, max write time: 20 ms (=200.0000 KB/s), fsync: 5 ms
INFO    Run 29:   280.14 KB/s, max write time: 266 ms (=  15.04 KB/s), fsync: 6 ms
INFO    Run 30:   267.48 KB/s, max write time: 275 ms (=  14.55 KB/s), fsync: 29 ms
INFO    Run 31:   355.11 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 5 ms
INFO    Run 32:   360.28 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 5 ms
INFO    Run 33:   361.13 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run 34:   361.91 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 5 ms
INFO    Run 35:   258.39 KB/s, max write time: 278 ms (=  14.39 KB/s), fsync: 5 ms
INFO    Run 36:   361.40 KB/s, max write time: 24 ms (= 166.67 KB/s), fsync: 5 ms
INFO    Run 37:   361.89 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run 38:   362.08 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 4 ms
INFO    Run 39:   362.14 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 5 ms
INFO    Run 40:   359.92 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 5 ms
INFO    Run 41:   333.03 KB/s, max write time: 183 ms (=  21.86 KB/s), fsync: 5 ms
INFO    Run 42:   340.65 KB/s, max write time: 136 ms (=  29.41 KB/s), fsync: 4 ms
INFO    Run 43:   361.62 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 6 ms
INFO    Run 44:   364.32 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run 45:   297.40 KB/s, max write time: 253 ms (=  15.81 KB/s), fsync: 29 ms
INFO    Run 46:   230.86 KB/s, max write time: 402 ms (=   9.95 KB/s), fsync: 6 ms
INFO    Run 47:   361.15 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Run 48:   358.03 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 6 ms
INFO    Run 49:   361.22 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO    Avg   :   334.20 KB/s

pkocmoud, 2016-05-28:
**Sandisk Ultra PLUS 16GB Class 10 SD card, PixRacer**

sd_bench -r 50
INFO  Using block size = 4096 bytes, sync=0
INFO
INFO  Testing Sequential Write Speed...
INFO    Run  0:   186.60 KB/s, max write time: 79 ms (=  50.63 KB/s), fsync: 21 ms
INFO    Run  1:   131.29 KB/s, max write time: 607 ms (=   6.59 KB/s), fsync: 305 ms
INFO    Run  2:   165.42 KB/s, max write time: 315 ms (=  12.70 KB/s), fsync: 28 ms
INFO    Run  3:   200.39 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 9 ms
INFO    Run  4:   199.56 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run  5:   201.60 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 9 ms
INFO    Run  6:   199.99 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run  7:   200.93 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run  8:   198.91 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run  9:   199.33 KB/s, max write time: 41 ms (=  97.56 KB/s), fsync: 9 ms
INFO    Run 10:   201.32 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 9 ms
INFO    Run 11:   202.39 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run 12:   199.65 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 10 ms
INFO    Run 13:   201.21 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run 14:   197.83 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 24 ms
INFO    Run 15:   201.73 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 8 ms
INFO    Run 16:   199.79 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 8 ms
INFO    Run 17:   200.74 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run 18:   200.21 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 11 ms
INFO    Run 19:   199.26 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run 20:   200.01 KB/s, max write time: 42 ms (=  95.24 KB/s), fsync: 9 ms
INFO    Run 21:   199.59 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 10 ms
INFO    Run 22:   199.90 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run 23:   200.97 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run 24:   200.46 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 10 ms
INFO    Run 25:   199.35 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 10 ms
INFO    Run 26:   200.71 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 9 ms
INFO    Run 27:   200.13 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 9 ms
INFO    Run 28:   200.10 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 9 ms
INFO    Run 29:   194.32 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 8 ms
INFO    Run 30:   194.52 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 38 ms
INFO    Run 31:   199.26 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run 32:   197.96 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 9 ms
INFO    Run 33:   199.21 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run 34:   197.84 KB/s, max write time: 39 ms (= 102.56 KB/s), fsync: 9 ms
INFO    Run 35:   197.50 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 9 ms
INFO    Run 36:   198.10 KB/s, max write time: 33 ms (= 121.21 KB/s), fsync: 11 ms
INFO    Run 37:   199.26 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 9 ms
INFO    Run 38:   198.89 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 9 ms
INFO    Run 39:   195.50 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 21 ms
INFO    Run 40:   199.27 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 9 ms
INFO    Run 41:   197.93 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 9 ms
INFO    Run 42:   198.60 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 10 ms
INFO    Run 43:   198.26 KB/s, max write time: 31 ms (= 129.03 KB/s), fsync: 10 ms
INFO    Run 44:   195.80 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 27 ms
INFO    Run 45:   199.17 KB/s, max write time: 33 ms (= 121.21 KB/s), fsync: 8 ms
INFO    Run 46:   198.10 KB/s, max write time: 33 ms (= 121.21 KB/s), fsync: 9 ms
INFO    Run 47:   198.52 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 9 ms
INFO    Run 48:   197.77 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 8 ms
INFO    Run 49:   196.77 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 8 ms
INFO    Avg   :   196.27 KB/s

pkocmoud, 2016-05-28:
**Sandisk Ultra 8GB Class 10 SD card, PixRacer**

sd_bench -r 50
INFO  Using block size = 4096 bytes, sync=0
INFO
INFO  Testing Sequential Write Speed...
INFO    Run  0:   350.26 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 8 ms
INFO    Run  1:   353.02 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 6 ms
INFO    Run  2:   355.78 KB/s, max write time: 39 ms (= 102.56 KB/s), fsync: 7 ms
INFO    Run  3:   352.80 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 6 ms
INFO    Run  4:   350.26 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 6 ms
INFO    Run  5:   350.47 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 5 ms
INFO    Run  6:   353.90 KB/s, max write time: 38 ms (= 105.26 KB/s), fsync: 14 ms
INFO    Run  7:   359.77 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 14 ms
INFO    Run  8:   357.11 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 7 ms
INFO    Run  9:   347.69 KB/s, max write time: 41 ms (=  97.56 KB/s), fsync: 8 ms
INFO    Run 10:   352.59 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 6 ms
INFO    Run 11:   352.02 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 8 ms
INFO    Run 12:   355.84 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 9 ms
INFO    Run 13:   353.09 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 8 ms
INFO    Run 14:   347.66 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 6 ms
INFO    Run 15:   343.07 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 8 ms
INFO    Run 16:   345.12 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 8 ms
INFO    Run 17:   342.12 KB/s, max write time: 41 ms (=  97.56 KB/s), fsync: 9 ms
INFO    Run 18:   347.27 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 9 ms
INFO    Run 19:   350.40 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 8 ms
INFO    Run 20:   348.97 KB/s, max write time: 27 ms (= 148.15 KB/s), fsync: 5 ms
INFO    Run 21:   344.03 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 9 ms
INFO    Run 22:   344.27 KB/s, max write time: 40 ms (=100.0000 KB/s), fsync: 5 ms
INFO    Run 23:   344.11 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 9 ms
INFO    Run 24:   350.57 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 10 ms
INFO    Run 25:   349.45 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 8 ms
INFO    Run 26:   346.11 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 8 ms
INFO    Run 27:   345.20 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 10 ms
INFO    Run 28:   346.29 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 8 ms
INFO    Run 29:   342.43 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 8 ms
INFO    Run 30:   346.95 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 27 ms
INFO    Run 31:   345.73 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 8 ms
INFO    Run 32:   346.35 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 8 ms
INFO    Run 33:   346.40 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 9 ms
INFO    Run 34:   344.85 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 5 ms
INFO    Run 35:   341.06 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 5 ms
INFO    Run 36:   352.94 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 6 ms
INFO    Run 37:   339.20 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 8 ms
INFO    Run 38:   345.98 KB/s, max write time: 27 ms (= 148.15 KB/s), fsync: 10 ms
INFO    Run 39:   344.35 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 6 ms
INFO    Run 40:   346.59 KB/s, max write time: 30 ms (= 133.33 KB/s), fsync: 5 ms
INFO    Run 41:   347.67 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 8 ms
INFO    Run 42:   347.25 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 5 ms
INFO    Run 43:   349.02 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 5 ms
INFO    Run 44:   347.11 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 9 ms
INFO    Run 45:   347.13 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 7 ms
INFO    Run 46:   338.37 KB/s, max write time: 36 ms (= 111.11 KB/s), fsync: 5 ms
INFO    Run 47:   352.09 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 9 ms
INFO    Run 48:   345.92 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 9 ms
INFO    Run 49:   344.64 KB/s, max write time: 33 ms (= 121.21 KB/s), fsync: 7 ms
INFO    Avg   :   348.03 KB/s

bkueng, 2016-07-15:
Got some new SD Cards to test. I used Pixracer, on current master:

**SanDisk Extreme U3 32GB**
```
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench]
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   457.55 KB/s, max write time: 16 ms (=250.000 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  1:   463.05 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  2:   460.20 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  3:   466.76 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  4:   460.39 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  5:   462.55 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  6:   464.69 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  7:   462.83 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  8:   462.44 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  9:   454.54 KB/s, max write time: 16 ms (=250.000 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 10:   459.93 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 11:   461.44 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 12:   464.18 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 13:   462.75 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 14:   458.16 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 15:   467.86 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 16:   463.20 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 17:   460.82 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 18:   459.05 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 19:   460.76 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 20:   463.27 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 21:   464.31 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 22:   457.38 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 23:   466.19 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 24:   463.08 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 25:   458.81 KB/s, max write time: 16 ms (=250.000 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 26:   460.69 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 27:   458.82 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 28:   462.36 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 29:   462.44 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 30:   459.85 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 31:   464.75 KB/s, max write time: 16 ms (=250.000 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 32:   462.29 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 33:   456.29 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 34:   462.79 KB/s, max write time: 16 ms (=250.000 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 35:   460.54 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 36:   462.59 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 37:   458.98 KB/s, max write time: 16 ms (=250.000 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 38:   464.12 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 39:   459.83 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 40:   462.08 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 41:   462.61 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 42:   460.17 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 43:   463.53 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 44:   462.72 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 45:   460.16 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 46:   464.25 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 47:   459.10 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 48:   464.12 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 49:   457.92 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 4 ms
INFO  [sd_bench]   Avg   :   461.58 KB/s
```

**SanDisk 32 GB Class 10 (High Endurance Video Monitoring Card)**
```
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench]
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   370.20 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  1:   350.33 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  2:   352.92 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 5 ms
[... 85 lines cut ...]
[... cut ...]
INFO  [sd_bench]   Run 47:   203.21 KB/s, max write time: 120 ms (=  33.33 KB/s), fsync: 11 ms
INFO  [sd_bench]   Run 48:   218.02 KB/s, max write time: 53 ms (=  75.47 KB/s), fsync: 43 ms
INFO  [sd_bench]   Run 49:   218.96 KB/s, max write time: 79 ms (=  50.63 KB/s), fsync: 11 ms
INFO  [sd_bench]   Avg   :   208.76 KB/s
```

The **SanDisk Extreme U3 32GB** is definitely the best we tested so far. It doesn't show the spikes we see on other cards, and the logger performs really well. Let's see what happens once it fills up.

kd0aij, 2016-08-08:
**Sandisk Extreme PLUS 16GB Class 10 SD card, PixRacer**
```
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   317.55 KB/s, max write time: 273 ms (=  14.65 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  1:   345.78 KB/s, max write time: 109 ms (=  36.70 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  2:   350.16 KB/s, max write time: 172 ms (=  23.26 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  3:   277.62 KB/s, max write time: 361 ms (=  11.08 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  4:   351.89 KB/s, max write time: 181 ms (=  22.10 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  5:   361.25 KB/s, max write time: 269 ms (=  14.87 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  6:   304.32 KB/s, max write time: 231 ms (=  17.32 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  7:   388.49 KB/s, max write time: 74 ms (=  54.05 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  8:   356.16 KB/s, max write time: 252 ms (=  15.87 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  9:   342.70 KB/s, max write time: 266 ms (=  15.04 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 10:   341.25 KB/s, max write time: 262 ms (=  15.27 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 11:   302.04 KB/s, max write time: 226 ms (=  17.70 KB/s), fsync: 182 ms
INFO  [sd_bench]   Run 12:   377.95 KB/s, max write time: 277 ms (=  14.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 13:   369.69 KB/s, max write time: 147 ms (=  27.21 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 14:   281.19 KB/s, max write time: 363 ms (=  11.02 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 15:   359.09 KB/s, max write time: 224 ms (=  17.86 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 16:   349.42 KB/s, max write time: 232 ms (=  17.24 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 17:   309.23 KB/s, max write time: 265 ms (=  15.09 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 18:   361.47 KB/s, max write time: 181 ms (=  22.10 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 19:   366.36 KB/s, max write time: 160 ms (=  25.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 20:   342.57 KB/s, max write time: 233 ms (=  17.17 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 21:   372.18 KB/s, max write time: 224 ms (=  17.86 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 22:   293.65 KB/s, max write time: 445 ms (=   8.99 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 23:   322.14 KB/s, max write time: 219 ms (=  18.26 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 24:   352.24 KB/s, max write time: 234 ms (=  17.09 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 25:   320.77 KB/s, max write time: 229 ms (=  17.47 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 26:   354.22 KB/s, max write time: 132 ms (=  30.30 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 27:   333.98 KB/s, max write time: 217 ms (=  18.43 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 28:   299.74 KB/s, max write time: 226 ms (=  17.70 KB/s), fsync: 54 ms
INFO  [sd_bench]   Run 29:   336.82 KB/s, max write time: 139 ms (=  28.78 KB/s), fsync: 3 ms
INFO  [sd_bench]   Run 30:   341.75 KB/s, max write time: 197 ms (=  20.30 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 31:   321.10 KB/s, max write time: 198 ms (=  20.20 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 32:   280.26 KB/s, max write time: 287 ms (=  13.94 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 33:   370.64 KB/s, max write time: 224 ms (=  17.86 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 34:   344.73 KB/s, max write time: 273 ms (=  14.65 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 35:   302.25 KB/s, max write time: 267 ms (=  14.98 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 36:   366.61 KB/s, max write time: 275 ms (=  14.55 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 37:   362.07 KB/s, max write time: 144 ms (=  27.78 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 38:   319.08 KB/s, max write time: 274 ms (=  14.60 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 39:   373.09 KB/s, max write time: 223 ms (=  17.94 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 40:   299.68 KB/s, max write time: 274 ms (=  14.60 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 41:   365.17 KB/s, max write time: 146 ms (=  27.40 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 42:   328.15 KB/s, max write time: 353 ms (=  11.33 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 43:   312.01 KB/s, max write time: 265 ms (=  15.09 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 44:   393.30 KB/s, max write time: 116 ms (=  34.48 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 45:   358.87 KB/s, max write time: 266 ms (=  15.04 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 46:   324.62 KB/s, max write time: 273 ms (=  14.65 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 47:   328.25 KB/s, max write time: 379 ms (=  10.55 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 48:   353.00 KB/s, max write time: 231 ms (=  17.32 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 49:   328.39 KB/s, max write time: 267 ms (=  14.98 KB/s), fsync: 4 ms
INFO  [sd_bench]   Avg   :   338.03 KB/s
```

kd0aij, 2016-08-08:
**note that the max write time increases dramatically on run 12**

**brand new Sandisk Extreme PLUS 16GB Class 10 SD card, PixRacer**
```
sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   381.14 KB/s, max write time: 16 ms (=250.000 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  1:   375.26 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  2:   372.04 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  3:   374.14 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  4:   372.92 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  5:   367.65 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  6:   370.39 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  7:   370.79 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  8:   367.98 KB/s, max write time: 20 ms (=200.0000 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  9:   367.17 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 10:   363.43 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 11:   367.44 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 12:   289.55 KB/s, max write time: 173 ms (=  23.12 KB/s), fsync: 78 ms
INFO  [sd_bench]   Run 13:   326.53 KB/s, max write time: 234 ms (=  17.09 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 14:   324.90 KB/s, max write time: 233 ms (=  17.17 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 15:   327.01 KB/s, max write time: 152 ms (=  26.32 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 16:   285.12 KB/s, max write time: 236 ms (=  16.95 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 17:   325.08 KB/s, max write time: 175 ms (=  22.86 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 18:   325.84 KB/s, max write time: 236 ms (=  16.95 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 19:   324.91 KB/s, max write time: 234 ms (=  17.09 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 20:   286.39 KB/s, max write time: 176 ms (=  22.73 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 21:   324.67 KB/s, max write time: 240 ms (=  16.67 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 22:   324.70 KB/s, max write time: 238 ms (=  16.81 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 23:   349.37 KB/s, max write time: 233 ms (=  17.17 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 24:   247.56 KB/s, max write time: 491 ms (=   8.15 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 25:   324.95 KB/s, max write time: 158 ms (=  25.32 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 26:   324.99 KB/s, max write time: 234 ms (=  17.09 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 27:   324.40 KB/s, max write time: 234 ms (=  17.09 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 28:   279.47 KB/s, max write time: 164 ms (=  24.39 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 29:   316.33 KB/s, max write time: 273 ms (=  14.65 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 30:   322.55 KB/s, max write time: 171 ms (=  23.39 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 31:   320.70 KB/s, max write time: 261 ms (=  15.33 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 32:   283.68 KB/s, max write time: 278 ms (=  14.39 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 33:   326.23 KB/s, max write time: 168 ms (=  23.81 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 34:   325.02 KB/s, max write time: 244 ms (=  16.39 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 35:   330.93 KB/s, max write time: 203 ms (=  19.70 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 36:   249.25 KB/s, max write time: 473 ms (=   8.46 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 37:   333.81 KB/s, max write time: 203 ms (=  19.70 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 38:   328.55 KB/s, max write time: 227 ms (=  17.62 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 39:   328.14 KB/s, max write time: 232 ms (=  17.24 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 40:   282.25 KB/s, max write time: 198 ms (=  20.20 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 41:   318.42 KB/s, max write time: 275 ms (=  14.55 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 42:   332.39 KB/s, max write time: 197 ms (=  20.30 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 43:   326.58 KB/s, max write time: 225 ms (=  17.78 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 44:   282.68 KB/s, max write time: 241 ms (=  16.60 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 45:   316.77 KB/s, max write time: 237 ms (=  16.88 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 46:   315.59 KB/s, max write time: 184 ms (=  21.74 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 47:   323.61 KB/s, max write time: 252 ms (=  15.87 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 48:   262.18 KB/s, max write time: 448 ms (=   8.93 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 49:   334.76 KB/s, max write time: 192 ms (=  20.83 KB/s), fsync: 5 ms
INFO  [sd_bench]   Avg   :   325.71 KB/s
```

kd0aij, 2016-08-28:
closed awaiting further interest

closed 2016-08-28 by kd0aij

davids5, 2016-08-28:
@kd0aij - FYI:I just merged master Last Fri.

kd0aij, 2016-08-28:
Thanks. I'm waiting to see if anyone can reproduce the behavior I reported above: https://github.com/PX4/Firmware/issues/4634#issuecomment-238289584

muramura, 2022-07-10:
TOSHIBA 8GB Class 4 SDHC, Pixhawk 6x

NuttShell (NSH) NuttX-10.2.0
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1697.72 KB/s, max write time: 136 ms (=  29.41 KB/s), fsync: 20 ms
INFO  [sd_bench]   Run  1:  2055.34 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  2:  2137.13 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run  3:  2069.75 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run  4:  2126.05 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  5:  2127.34 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  6:  2126.21 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  7:  2115.59 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run  8:  2138.72 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run  9:  2065.96 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 10:  2155.68 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 11:  2125.42 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 12:  2138.96 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 13:  2148.35 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 14:  2136.90 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 15:  2123.26 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 16:  2132.34 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 17:  2140.00 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 18:  2156.59 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 19:  2122.83 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 20:  2167.41 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 21:   764.62 KB/s, max write time: 1163 ms (=   3.44 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 22:  2159.97 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 23:  2152.56 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 24:  2140.88 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 25:  2149.53 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 26:  2156.37 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 27:  1020.60 KB/s, max write time: 1161 ms (=   3.45 KB/s), fsync: 21 ms
INFO  [sd_bench]   Run 28:  2048.74 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 29:  2151.79 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 30:  2155.64 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 31:  2142.26 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 32:  2157.75 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 33:  1189.83 KB/s, max write time: 1164 ms (=   3.44 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 34:  2039.74 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 35:  2144.82 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 36:  2153.34 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 37:  2124.87 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 38:  2176.38 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 39:  1201.14 KB/s, max write time: 1161 ms (=   3.45 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 40:  2054.56 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 41:  2137.30 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 42:  2169.82 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 43:  2131.44 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 44:  2153.06 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 45:  1201.13 KB/s, max write time: 1157 ms (=   3.46 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 46:  2062.30 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 47:  2138.98 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 48:  2150.83 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 49:  2150.16 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 8 ms
INFO  [sd_bench]   Avg   :  1997.64 KB/s
nsh>

muramura, 2022-07-16:
SAMSUNG 128GB EVO PLUS

NuttShell (NSH) NuttX-10.2.0
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1220.30 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 12 ms
INFO  [sd_bench]   Run  1:  1220.30 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run  2:  1207.51 KB/s, max write time: 26 ms (= 153.85 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run  3:  1304.23 KB/s, max write time: 27 ms (= 148.15 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run  4:  1207.60 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run  5:  1233.29 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run  6:  1216.85 KB/s, max write time: 24 ms (= 166.67 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run  7:  1219.08 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 21 ms
INFO  [sd_bench]   Run  8:  1223.88 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 20 ms
INFO  [sd_bench]   Run  9:  1217.40 KB/s, max write time: 20 ms (= 200.00 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 10:  1212.05 KB/s, max write time: 26 ms (= 153.85 KB/s), fsync: 12 ms
INFO  [sd_bench]   Run 11:  1223.47 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 10 ms
INFO  [sd_bench]   Run 12:  1229.95 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 10 ms
INFO  [sd_bench]   Run 13:  1222.99 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 14:  1204.31 KB/s, max write time: 26 ms (= 153.85 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 15:  1227.42 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 16:  1222.24 KB/s, max write time: 20 ms (= 200.00 KB/s), fsync: 10 ms
INFO  [sd_bench]   Run 17:  1281.57 KB/s, max write time: 27 ms (= 148.15 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 18:  1211.90 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 19:  1230.96 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 20:  1231.19 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 10 ms
INFO  [sd_bench]   Run 21:  1207.47 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 22:  1225.67 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 23:  1203.11 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 20 ms
INFO  [sd_bench]   Run 24:  1223.97 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 25:  1209.87 KB/s, max write time: 23 ms (= 173.91 KB/s), fsync: 12 ms
INFO  [sd_bench]   Run 26:  1228.90 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 10 ms
INFO  [sd_bench]   Run 27:  1224.88 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 12 ms
INFO  [sd_bench]   Run 28:  1211.57 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 29:  1202.91 KB/s, max write time: 24 ms (= 166.67 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 30:  1222.79 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 20 ms
INFO  [sd_bench]   Run 31:  1241.80 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 32:  1283.24 KB/s, max write time: 24 ms (= 166.67 KB/s), fsync: 12 ms
INFO  [sd_bench]   Run 33:  1195.00 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 34:  1221.66 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 20 ms
INFO  [sd_bench]   Run 35:  1224.21 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 12 ms
INFO  [sd_bench]   Run 36:  1213.14 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 37:  1229.97 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 14 ms
INFO  [sd_bench]   Run 38:  1216.09 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 11 ms
INFO  [sd_bench]   Run 39:  1224.41 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 10 ms
INFO  [sd_bench]   Run 40:  1213.53 KB/s, max write time: 23 ms (= 173.91 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 41:  1226.17 KB/s, max write time: 20 ms (= 200.00 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 42:  1224.68 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 43:  1206.52 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 10 ms
INFO  [sd_bench]   Run 44:  1213.68 KB/s, max write time: 26 ms (= 153.85 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 45:  1224.41 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 46:  1300.56 KB/s, max write time: 24 ms (= 166.67 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 47:  1210.37 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 20 ms
INFO  [sd_bench]   Run 48:  1215.63 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 12 ms
INFO  [sd_bench]   Run 49:  1221.43 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 15 ms
INFO  [sd_bench]   Avg   :  1224.72 KB/s
nsh>

muramura, 2022-07-31:
Transcend 4GB Class 4 SDHC, Pixhawk 6x

NuttShell (NSH) NuttX-10.2.0
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   796.19 KB/s, max write time: 50 ms (=  80.00 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run  1:   800.64 KB/s, max write time: 44 ms (=  90.91 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run  2:   874.62 KB/s, max write time: 66 ms (=  60.61 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run  3:   809.18 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run  4:   808.30 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run  5:   824.94 KB/s, max write time: 59 ms (=  67.80 KB/s), fsync: 21 ms
INFO  [sd_bench]   Run  6:   834.60 KB/s, max write time: 69 ms (=  57.97 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run  7:   837.61 KB/s, max write time: 55 ms (=  72.73 KB/s), fsync: 27 ms
INFO  [sd_bench]   Run  8:   828.62 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run  9:   878.62 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 43 ms
INFO  [sd_bench]   Run 10:   861.31 KB/s, max write time: 84 ms (=  47.62 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 11:   844.95 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 12:   860.89 KB/s, max write time: 75 ms (=  53.33 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 13:   843.35 KB/s, max write time: 83 ms (=  48.19 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run 14:   899.39 KB/s, max write time: 76 ms (=  52.63 KB/s), fsync: 24 ms
INFO  [sd_bench]   Run 15:   836.15 KB/s, max write time: 64 ms (=  62.50 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 16:   910.50 KB/s, max write time: 73 ms (=  54.79 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 17:   844.05 KB/s, max write time: 84 ms (=  47.62 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 18:   897.03 KB/s, max write time: 49 ms (=  81.63 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run 19:   845.99 KB/s, max write time: 66 ms (=  60.61 KB/s), fsync: 21 ms
INFO  [sd_bench]   Run 20:   874.43 KB/s, max write time: 57 ms (=  70.18 KB/s), fsync: 54 ms
INFO  [sd_bench]   Run 21:   860.97 KB/s, max write time: 76 ms (=  52.63 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 22:   867.41 KB/s, max write time: 81 ms (=  49.38 KB/s), fsync: 23 ms
INFO  [sd_bench]   Run 23:   885.99 KB/s, max write time: 81 ms (=  49.38 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run 24:   788.65 KB/s, max write time: 69 ms (=  57.97 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 25:   835.52 KB/s, max write time: 76 ms (=  52.63 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 26:   805.91 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 53 ms
INFO  [sd_bench]   Run 27:   843.81 KB/s, max write time: 70 ms (=  57.14 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 28:   796.27 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 29:   877.42 KB/s, max write time: 78 ms (=  51.28 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run 30:   863.21 KB/s, max write time: 52 ms (=  76.92 KB/s), fsync: 50 ms
INFO  [sd_bench]   Run 31:   851.07 KB/s, max write time: 75 ms (=  53.33 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 32:   872.85 KB/s, max write time: 42 ms (=  95.24 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 33:   829.83 KB/s, max write time: 88 ms (=  45.45 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 34:   897.01 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 43 ms
INFO  [sd_bench]   Run 35:   848.00 KB/s, max write time: 69 ms (=  57.97 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 36:   837.53 KB/s, max write time: 54 ms (=  74.07 KB/s), fsync: 80 ms
INFO  [sd_bench]   Run 37:   849.38 KB/s, max write time: 75 ms (=  53.33 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 38:   844.25 KB/s, max write time: 54 ms (=  74.07 KB/s), fsync: 43 ms
INFO  [sd_bench]   Run 39:   887.18 KB/s, max write time: 75 ms (=  53.33 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 40:   828.75 KB/s, max write time: 75 ms (=  53.33 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 41:   900.96 KB/s, max write time: 75 ms (=  53.33 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 42:   822.83 KB/s, max write time: 73 ms (=  54.79 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run 43:   883.38 KB/s, max write time: 69 ms (=  57.97 KB/s), fsync: 55 ms
INFO  [sd_bench]   Run 44:   793.52 KB/s, max write time: 62 ms (=  64.52 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 45:   794.37 KB/s, max write time: 74 ms (=  54.05 KB/s), fsync: 24 ms
INFO  [sd_bench]   Run 46:   820.24 KB/s, max write time: 61 ms (=  65.57 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run 47:   800.96 KB/s, max write time: 80 ms (=  50.00 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run 48:   750.62 KB/s, max write time: 79 ms (=  50.63 KB/s), fsync: 81 ms
INFO  [sd_bench]   Run 49:   811.74 KB/s, max write time: 54 ms (=  74.07 KB/s), fsync: 18 ms
INFO  [sd_bench]   Avg   :   843.20 KB/s
nsh>

muramura, 2022-08-11:
SANDISK 4GB Class 4 SDHC, Pixhawk 6C

NuttShell (NSH) NuttX-10.2.0
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  2251.45 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  1:  2252.02 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  2:  2242.82 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  3:  2250.08 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  4:  2252.01 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  5:  2254.64 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  6:  2258.82 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run  7:  2257.36 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  8:  2241.24 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run  9:  2249.13 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 10:  2266.72 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 11:  2279.15 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 12:  2227.57 KB/s, max write time: 23 ms (= 173.91 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 13:  2248.67 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 14:  1886.80 KB/s, max write time: 170 ms (=  23.53 KB/s), fsync: 139 ms ★
INFO  [sd_bench]   Run 15:  2263.24 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 16:  2256.28 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 17:  2269.82 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 18:  2252.05 KB/s, max write time: 21 ms (= 190.48 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 19:  2268.90 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 20:  2266.35 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 21:  2265.90 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 22:  2240.78 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 23:  2270.19 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 24:  2246.27 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 25:  2272.51 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 26:  2255.92 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 27:  2261.71 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 28:  2259.95 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 29:  2270.34 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 30:  2266.95 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 31:  2245.75 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 32:  2270.29 KB/s, max write time: 18 ms (= 222.22 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 33:  2262.43 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 34:  2249.21 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 35:  2254.95 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 36:  2263.49 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 10 ms
INFO  [sd_bench]   Run 37:  2289.22 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 38:  2255.04 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 39:  2255.03 KB/s, max write time: 14 ms (= 285.71 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 40:  2226.45 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 41:  2277.37 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 42:  2263.62 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 43:  1857.35 KB/s, max write time: 171 ms (=  23.39 KB/s), fsync: 138 ms ★
INFO  [sd_bench]   Run 44:  2239.42 KB/s, max write time: 17 ms (= 235.29 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 45:  2264.23 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 46:  2268.54 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 47:  2272.09 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 48:  2260.04 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 49:  2255.91 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 4 ms
INFO  [sd_bench]   Avg   :  2241.01 KB/s
nsh>

muramura, 2022-08-11:
SP 4GB Class 10 SDHC, Pixhawk 6C

NuttShell (NSH) NuttX-10.2.0
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   701.76 KB/s, max write time: 407 ms (=   9.83 KB/s), fsync: 278 ms ★
INFO  [sd_bench]   Run  1:   701.63 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 11 ms ★
INFO  [sd_bench]   Run  2:   557.67 KB/s, max write time: 416 ms (=   9.62 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run  3:   620.52 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 11 ms ★
INFO  [sd_bench]   Run  4:   792.57 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 11 ms ★
INFO  [sd_bench]   Run  5:   513.34 KB/s, max write time: 447 ms (=   8.95 KB/s), fsync: 9 ms ★
INFO  [sd_bench]   Run  6:   622.45 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 11 ms ★
INFO  [sd_bench]   Run  7:   770.07 KB/s, max write time: 410 ms (=   9.76 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run  8:   533.84 KB/s, max write time: 412 ms (=   9.71 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run  9:   623.10 KB/s, max write time: 407 ms (=   9.83 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 10:   774.36 KB/s, max write time: 419 ms (=   9.55 KB/s), fsync: 11 ms ★
INFO  [sd_bench]   Run 11:   529.03 KB/s, max write time: 413 ms (=   9.69 KB/s), fsync: 9 ms ★
INFO  [sd_bench]   Run 12:   623.01 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 11 ms ★
INFO  [sd_bench]   Run 13:   773.50 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 11 ms ★
INFO  [sd_bench]   Run 14:   532.09 KB/s, max write time: 411 ms (=   9.73 KB/s), fsync: 9 ms ★
INFO  [sd_bench]   Run 15:   619.34 KB/s, max write time: 407 ms (=   9.83 KB/s), fsync: 22 ms ★
INFO  [sd_bench]   Run 16:   782.21 KB/s, max write time: 416 ms (=   9.62 KB/s), fsync: 9 ms ★
INFO  [sd_bench]   Run 17:   510.71 KB/s, max write time: 410 ms (=   9.76 KB/s), fsync: 72 ms ★
INFO  [sd_bench]   Run 18:   617.01 KB/s, max write time: 416 ms (=   9.62 KB/s), fsync: 17 ms ★
INFO  [sd_bench]   Run 19:   761.23 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 73 ms ★
INFO  [sd_bench]   Run 20:   522.93 KB/s, max write time: 412 ms (=   9.71 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 21:   605.11 KB/s, max write time: 419 ms (=   9.55 KB/s), fsync: 68 ms ★
INFO  [sd_bench]   Run 22:   772.68 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 68 ms ★
INFO  [sd_bench]   Run 23:   516.67 KB/s, max write time: 411 ms (=   9.73 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 24:   603.59 KB/s, max write time: 407 ms (=   9.83 KB/s), fsync: 70 ms ★
INFO  [sd_bench]   Run 25:   783.02 KB/s, max write time: 409 ms (=   9.78 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 26:   508.42 KB/s, max write time: 413 ms (=   9.69 KB/s), fsync: 9 ms ★
INFO  [sd_bench]   Run 27:   621.65 KB/s, max write time: 409 ms (=   9.78 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 28:   758.69 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 71 ms ★
INFO  [sd_bench]   Run 29:   522.89 KB/s, max write time: 436 ms (=   9.17 KB/s), fsync: 9 ms ★
INFO  [sd_bench]   Run 30:   605.70 KB/s, max write time: 409 ms (=   9.78 KB/s), fsync: 68 ms ★
INFO  [sd_bench]   Run 31:   768.59 KB/s, max write time: 410 ms (=   9.76 KB/s), fsync: 68 ms ★
INFO  [sd_bench]   Run 32:   515.77 KB/s, max write time: 416 ms (=   9.62 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 33:   609.09 KB/s, max write time: 407 ms (=   9.83 KB/s), fsync: 67 ms ★
INFO  [sd_bench]   Run 34:   791.85 KB/s, max write time: 409 ms (=   9.78 KB/s), fsync: 9 ms ★
INFO  [sd_bench]   Run 35:   500.86 KB/s, max write time: 423 ms (=   9.46 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 36:   621.13 KB/s, max write time: 410 ms (=   9.76 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 37:   753.57 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 69 ms ★
INFO  [sd_bench]   Run 38:   513.29 KB/s, max write time: 410 ms (=   9.76 KB/s), fsync: 71 ms ★
INFO  [sd_bench]   Run 39:   624.12 KB/s, max write time: 414 ms (=   9.66 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 40:   759.31 KB/s, max write time: 418 ms (=   9.57 KB/s), fsync: 69 ms ★
INFO  [sd_bench]   Run 41:   523.85 KB/s, max write time: 411 ms (=   9.73 KB/s), fsync: 9 ms ★
INFO  [sd_bench]   Run 42:   605.14 KB/s, max write time: 411 ms (=   9.73 KB/s), fsync: 69 ms ★
INFO  [sd_bench]   Run 43:   779.77 KB/s, max write time: 416 ms (=   9.62 KB/s), fsync: 9 ms ★
INFO  [sd_bench]   Run 44:   512.73 KB/s, max write time: 411 ms (=   9.73 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 45:   620.35 KB/s, max write time: 417 ms (=   9.59 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 46:   761.26 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 70 ms ★
INFO  [sd_bench]   Run 47:   525.56 KB/s, max write time: 413 ms (=   9.69 KB/s), fsync: 10 ms ★
INFO  [sd_bench]   Run 48:   604.02 KB/s, max write time: 420 ms (=   9.52 KB/s), fsync: 70 ms ★
INFO  [sd_bench]   Run 49:   767.42 KB/s, max write time: 408 ms (=   9.80 KB/s), fsync: 71 ms ★
INFO  [sd_bench]   Avg   :   636.96 KB/s

greinet, 2023-02-22:
**SanDisk Extreme microSDHC 3 Rescue Pro Deluxe Memory Card, Gold/Red 32GB ([Amazon](https://www.amazon.de/SanDisk-Extreme-microSDHC-Speicherkarte-SD-Adapter/dp/B06XWMQ81P)) on CubeOrange, PX4 Version 1.13.2**

```
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:   113.67 KB/s, max write time: 77 ms (=  51.95 KB/s), fsync: 24 ms
INFO  [sd_bench]   Run  1:   118.71 KB/s, max write time: 64 ms (=  62.50 KB/s), fsync: 11 ms
INFO  [sd_bench]   Run  2:   118.84 KB/s, max write time: 61 ms (=  65.57 KB/s), fsync: 51 ms
INFO  [sd_bench]   Run  3:   115.33 KB/s, max write time: 88 ms (=  45.45 KB/s), fsync: 59 ms
INFO  [sd_bench]   Run  4:   120.09 KB/s, max write time: 62 ms (=  64.52 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run  5:   115.76 KB/s, max write time: 61 ms (=  65.57 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run  6:   117.09 KB/s, max write time: 70 ms (=  57.14 KB/s), fsync: 41 ms
INFO  [sd_bench]   Run  7:   118.06 KB/s, max write time: 62 ms (=  64.52 KB/s), fsync: 47 ms
INFO  [sd_bench]   Run  8:   118.03 KB/s, max write time: 89 ms (=  44.94 KB/s), fsync: 20 ms
INFO  [sd_bench]   Run  9:   118.46 KB/s, max write time: 57 ms (=  70.18 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 10:   113.78 KB/s, max write time: 82 ms (=  48.78 KB/s), fsync: 21 ms
INFO  [sd_bench]   Run 11:   119.69 KB/s, max write time: 58 ms (=  68.97 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run 12:   118.51 KB/s, max write time: 59 ms (=  67.80 KB/s), fsync: 52 ms
INFO  [sd_bench]   Run 13:   116.37 KB/s, max write time: 88 ms (=  45.45 KB/s), fsync: 46 ms
INFO  [sd_bench]   Run 14:   116.20 KB/s, max write time: 66 ms (=  60.61 KB/s), fsync: 32 ms
INFO  [sd_bench]   Run 15:   118.26 KB/s, max write time: 88 ms (=  45.45 KB/s), fsync: 24 ms
INFO  [sd_bench]   Run 16:   118.93 KB/s, max write time: 70 ms (=  57.14 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 17:   116.89 KB/s, max write time: 73 ms (=  54.79 KB/s), fsync: 44 ms
INFO  [sd_bench]   Run 18:   118.49 KB/s, max write time: 81 ms (=  49.38 KB/s), fsync: 18 ms
INFO  [sd_bench]   Run 19:   118.05 KB/s, max write time: 62 ms (=  64.52 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 20:   114.25 KB/s, max write time: 104 ms (=  38.46 KB/s), fsync: 30 ms
INFO  [sd_bench]   Run 21:   119.93 KB/s, max write time: 64 ms (=  62.50 KB/s), fsync: 13 ms
INFO  [sd_bench]   Run 22:   120.34 KB/s, max write time: 70 ms (=  57.14 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 23:   114.55 KB/s, max write time: 82 ms (=  48.78 KB/s), fsync: 68 ms
INFO  [sd_bench]   Run 24:   116.70 KB/s, max write time: 73 ms (=  54.79 KB/s), fsync: 16 ms
INFO  [sd_bench]   Run 25:   115.86 KB/s, max write time: 77 ms (=  51.95 KB/s), fsync: 22 ms
INFO  [sd_bench]   Run 26:   118.31 KB/s, max write time: 61 ms (=  65.57 KB/s), fsync: 28 ms
INFO  [sd_bench]   Run 27:   117.59 KB/s, max write time: 70 ms (=  57.14 KB/s), fsync: 15 ms
INFO  [sd_bench]   Run 28:   118.07 KB/s, max write time: 68 ms (=  58.82 KB/s), fsync: 32 ms
INFO  [sd_bench]   Run 29:   118.95 KB/s, max write time: 61 ms (=  65.57 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 30:   118.32 KB/s, max write time: 60 ms (=  66.67 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run 31:   113.20 KB/s, max write time: 72 ms (=  55.56 KB/s), fsync: 11 ms
INFO  [sd_bench]   Run 32:   117.51 KB/s, max write time: 61 ms (=  65.57 KB/s), fsync: 38 ms
INFO  [sd_bench]   Run 33:   118.74 KB/s, max write time: 69 ms (=  57.97 KB/s), fsync: 13 ms
INFO  [sd_bench]   Run 34:   116.29 KB/s, max write time: 85 ms (=  47.06 KB/s), fsync: 21 ms
INFO  [sd_bench]   Run 35:   118.57 KB/s, max write time: 60 ms (=  66.67 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 36:   116.62 KB/s, max write time: 62 ms (=  64.52 KB/s), fsync: 11 ms
INFO  [sd_bench]   Run 37:   117.85 KB/s, max write time: 62 ms (=  64.52 KB/s), fsync: 19 ms
INFO  [sd_bench]   Run 38:   115.29 KB/s, max write time: 64 ms (=  62.50 KB/s), fsync: 22 ms
INFO  [sd_bench]   Run 39:   118.45 KB/s, max write time: 65 ms (=  61.54 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 40:   119.70 KB/s, max write time: 64 ms (=  62.50 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 41:   113.14 KB/s, max write time: 102 ms (=  39.22 KB/s), fsync: 47 ms
INFO  [sd_bench]   Run 42:   118.74 KB/s, max write time: 70 ms (=  57.14 KB/s), fsync: 33 ms
INFO  [sd_bench]   Run 43:   116.92 KB/s, max write time: 65 ms (=  61.54 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 44:   117.09 KB/s, max write time: 75 ms (=  53.33 KB/s), fsync: 37 ms
INFO  [sd_bench]   Run 45:   118.51 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 42 ms
INFO  [sd_bench]   Run 46:   117.74 KB/s, max write time: 89 ms (=  44.94 KB/s), fsync: 17 ms
INFO  [sd_bench]   Run 47:   117.97 KB/s, max write time: 62 ms (=  64.52 KB/s), fsync: 20 ms
INFO  [sd_bench]   Run 48:   114.27 KB/s, max write time: 81 ms (=  49.38 KB/s), fsync: 53 ms
INFO  [sd_bench]   Run 49:   120.57 KB/s, max write time: 63 ms (=  63.49 KB/s), fsync: 11 ms
INFO  [sd_bench]   Avg   :   117.38 KB/s
```
We tried to reproduce to results from @bkueng described [here](https://github.com/PX4/PX4-Autopilot/issues/4634#issuecomment-232935301) but were not able to. Any ideas why our speed is so much slower, is it a different card?

davids5, 2023-02-23:
@greinet - check and see if they used SDMMC1 or SDMMC2. Performance is much better on SDMMC2 due to DMA restrictions on  SDMMC1.

swimmingseeds, 2023-02-24:
This issue has been mentioned on **Discussion Forum for PX4, Pixhawk, QGroundControl, MAVSDK, MAVLink**. There might be relevant details there:

https://discuss.px4.io/t/sd-card-logging-performance-on-cube-orange/30966/1

wangwwno1, 2024-12-12:
**SanDisk Ultra Class 10 32GB on Pixhawk 6C Mini, PX4 Version 1.15.2**

```
NuttShell (NSH) NuttX-11.0.0
nsh> 
nsh> sd_bench -r 50
INFO  [sd_bench] Using block size = 4096 bytes, sync=0
INFO  [sd_bench] 
INFO  [sd_bench] Testing Sequential Write Speed...
INFO  [sd_bench]   Run  0:  1228.66 KB/s, max write time: 125 ms (=  32.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  1:  1855.19 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  2:  1768.82 KB/s, max write time: 24 ms (= 166.67 KB/s), fsync: 38 ms
INFO  [sd_bench]   Run  3:  1458.58 KB/s, max write time: 33 ms (= 121.21 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  4:  1857.32 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  5:  1260.71 KB/s, max write time: 123 ms (=  32.52 KB/s), fsync: 49 ms
INFO  [sd_bench]   Run  6:  1665.29 KB/s, max write time: 33 ms (= 121.21 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run  7:  1848.48 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run  8:  1835.31 KB/s, max write time: 22 ms (= 181.82 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run  9:  1825.20 KB/s, max write time: 39 ms (= 102.56 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 10:  1883.16 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 11:  1188.95 KB/s, max write time: 101 ms (=  39.60 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 12:  1852.68 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 13:  1682.41 KB/s, max write time: 34 ms (= 117.65 KB/s), fsync: 52 ms
INFO  [sd_bench]   Run 14:  1862.68 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 15:  1863.40 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 16:  1551.44 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 17:  1836.09 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 18:  1846.78 KB/s, max write time: 10 ms (= 400.00 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 19:   370.88 KB/s, max write time: 131 ms (=  30.53 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 20:  1833.93 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 1 ms
INFO  [sd_bench]   Run 21:  1824.84 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 22:  1436.20 KB/s, max write time: 51 ms (=  78.43 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 23:  1885.40 KB/s, max write time: 8 ms (= 500.00 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 24:  1875.98 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 25:  1457.93 KB/s, max write time: 85 ms (=  47.06 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 26:  1848.87 KB/s, max write time: 16 ms (= 250.00 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 27:  1821.64 KB/s, max write time: 32 ms (= 125.00 KB/s), fsync: 1 ms
INFO  [sd_bench]   Run 28:  1868.44 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 8 ms
INFO  [sd_bench]   Run 29:  1846.76 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 30:  1827.09 KB/s, max write time: 24 ms (= 166.67 KB/s), fsync: 9 ms
INFO  [sd_bench]   Run 31:  1864.84 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 32:  1821.34 KB/s, max write time: 28 ms (= 142.86 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 33:  1858.43 KB/s, max write time: 12 ms (= 333.33 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 34:  1832.15 KB/s, max write time: 15 ms (= 266.67 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 35:  1835.12 KB/s, max write time: 43 ms (=  93.02 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 36:  1888.11 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 37:  1746.49 KB/s, max write time: 40 ms (= 100.00 KB/s), fsync: 48 ms
INFO  [sd_bench]   Run 38:   784.05 KB/s, max write time: 127 ms (=  31.50 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 39:  1824.30 KB/s, max write time: 19 ms (= 210.53 KB/s), fsync: 7 ms
INFO  [sd_bench]   Run 40:  1783.66 KB/s, max write time: 46 ms (=  86.96 KB/s), fsync: 42 ms
INFO  [sd_bench]   Run 41:  1575.09 KB/s, max write time: 55 ms (=  72.73 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 42:  1838.29 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 43:  1485.55 KB/s, max write time: 39 ms (= 102.56 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 44:  1855.30 KB/s, max write time: 13 ms (= 307.69 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 45:  1833.93 KB/s, max write time: 9 ms (= 444.44 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 46:  1220.74 KB/s, max write time: 112 ms (=  35.71 KB/s), fsync: 4 ms
INFO  [sd_bench]   Run 47:  1854.83 KB/s, max write time: 11 ms (= 363.64 KB/s), fsync: 6 ms
INFO  [sd_bench]   Run 48:  1820.48 KB/s, max write time: 41 ms (=  97.56 KB/s), fsync: 5 ms
INFO  [sd_bench]   Run 49:  1561.02 KB/s, max write time: 29 ms (= 137.93 KB/s), fsync: 4 ms
INFO  [sd_bench]   Avg   :  1686.83 KB/s
INFO  [sd_bench]   Overall max write time: 131 ms
```