PX4/PX4-Autopilot #4635: sf0x driver reported not to be working

julianoes, 2016-05-25:
There was a report that the sf0x driver is not working anymore.

https://groups.google.com/forum/#!searchin/px4users/sf0x$20drivers$20not$20working/px4users/r0En9KarQr0/If4758M6AgAJ

LorenzMeier, 2016-05-25:
Reported as actually working, user error.

closed 2016-05-25 by LorenzMeier

weiminshen99, 2018-07-05:
@LorenzMeier 
I am testing a new sensor SF10/a from lightware.co.za, and I have connect the sensor to SERIAL/4, and set SENS_EN_SF0X = 2, but it does not seem working properly and nsh shows that all reading from the sensor are errors (see below). Could you please tell me what I did wrong? Thanks!
Wingman

===================
> python mavlink_shell.py 
Connecting to MAVLINK...

nsh> sf0x info
state @ 1000b0e0
sf0x_read: 10908 events, 94312us elapsed, 8us avg, min 6us max 318us 14.305us rms
sf0x_com_err: 10908 events
poll interval:  31 ticks
report queue	2/64 (2/2 @ 1000b1a0)

nsh> sf0x stop

nsh> sf0x start

nsh> sf0x info
state @ 1000b0e0
sf0x_read: 230 events, 1879us elapsed, 8us avg, min 6us max 111us 9.146us rms
sf0x_com_err: 231 events
poll interval:  31 ticks
report queue	2/64 (2/2 @ 20025690)
nsh>