PX4/PX4-Autopilot #4644: Pixracer out of memory

birchera, 2016-05-26:
While testing EKF2 with pixracer on a fixed-wing, we encountered problems with memory. Running all the standard modules and only 7kB stack for logging, we were not able to do any calibrations, due to the lack of memory. Disabling the position controller allowed us to do the calibration. However, when re-enabled the position controller, some modules also don't run correctly during flight.

[image]

FYI @tumbili @priseborough @CarlOlsson

julianoes, 2016-05-26:
Nothing seems excessive except that 4 mavlink instances are running. Are they all needed?

birchera, 2016-05-27:
I made a couple more tests and it turns out that every time I flash (ea138cfd380c06179d2e89cdd8167637dbd96d54 with EKF2 enabled), different errors occur:
* Pixracer is spontaneously rebooting, every other 30s
* Cutecom connection is suddenly closing and cannot be reopened until rebooting
* Calibration is not possible, reason given in QGC: Out of memory
* Cutecom connection is closing upon calibration
Without reflashing, errors seemed repeatable so far.

When starting with the old estimator, none of these have occurred.

julianoes, 2016-05-27:
@birchera: interesting. Don't you get a stack trace when Pixracer reboots?

LorenzMeier, 2016-05-28:
@priseborough This is something worthwhile to keep an eye on. It could point to stack smashing or out of mem.

LorenzMeier, 2016-05-28:
This is the right command to hunt some of that down: ```arm-none-eabi-nm --size-sort --print-size --radix=dec build_px4fmu-v2_default/src/firmware/nuttx/firmware_nuttx | grep " [bBdD] "```

LorenzMeier, 2016-05-28:
Related issue:
https://github.com/PX4/Firmware/issues/4677

If you disable UAVCAN and the Snapdragon drivers in the build you should be able to free up 3K.

LorenzMeier, 2016-05-30:
@birchera I should have fixed this on master. Can you re-try?

birchera, 2016-05-31:
Thanks Lorenz! On fixed-wing, pixracer, EKF2, we have now ~26K free memory, which should be sufficient.

closed 2016-05-31 by birchera

tumbili, 2016-05-31:
@birchera  what size  of  logbuffer  are  you  using  now?

birchera, 2016-05-31:
Standard 12K