PX4/PX4-Autopilot #4650: servo center adjustment

dagar, 2016-05-26:
It would be nice to be able to tweak the servo center slightly at runtime. Sometimes it's hard to center servos perfectly, especially field replacements.

Could we somehow modify the active mixer, use params, mavlink, etc? I think a simple param for each servo channel would get the job done, but correcting the mixer might make more sense?

tubeme, 2016-05-26:
Very good idea!

edsellar, 2016-05-30:
I've run into this issue so many times. Definitely a great idea!

kd0aij, 2016-05-30:
I vote for exposing the mixer parameters to allow setting center *and endpoints* to provide the same functionality that one normally has with an RC transmitter.

What is "mavlink ftp"? Would you copy a new mixer into RAM, then invoke "mixer load"? That sounds pretty awkward to implement, since you'd have to edit a mixer file somewhere first.

dagar, 2016-05-30:
The other piece of the mavlink ftp idea is a simple editor in QGC and one
button to transfer and load the new mixer.

dagar, 2016-05-30:
Yet another idea - onboard calibration.
With the basic mixer in place you could be walked through each output and asked to hold center and then each extreme.

kd0aij, 2016-05-30:
It appears that the servo center point is implicitly the midpoint between pwm_min and pwm_max:
https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/pwm_limit/pwm_limit.c#L181

That would make the center point dependent on the endpoints.

LorenzMeier, 2016-05-30:
I think we should introduce parameters (like the reverse params) for this purpose. Best would be to use the mixer, but we should take one step at a time.

dagar, 2016-05-30:
Once we have that in place we could add a simple setup screen in QGC using the metadata from the airframe.
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/2101_fw_AERT#L7

LorenzMeier, 2016-05-30:
Let me handle that here. Your time is way better spent with CI. I know what I have to do here.

dagar, 2016-05-30:
Sounds good to me!
Thanks Lorenz

dagar, 2016-10-16:
@bartosz-wawrzacz implementation here - https://github.com/PX4/Firmware/issues/4650

dagar, 2016-11-08:
https://github.com/PX4/Firmware/pull/5720

closed 2016-11-08 by dagar