PX4/PX4-Autopilot #4652: Snapdragon flight always use internal compass despite I set SENS_EXT_MAG to "External is primary Mag"

SamChenzx, 2016-05-27:
I've already used 3DR GPS/MAG module as the external GPS/Mag. Today I find out if I rotate the GPS module, heading will not change on QGC, but if snapdragon flight is rotated, heading changes. SENS_EXT_MAG is set to "External is primary Mag" but seems didn't work. Besides, without GPS lock, heading will slowly turn at random directions. In a real flight this afternoon, I tested POSCTL and Stabilized flight mode, both will takeoff but in the same time spin around fast, just like I continuously give quadrotor a yaw control. In the bench test, I find out one the output PWM duty of the forth motor is lower then other motors though I just add little throttle, without roll, pitch or yaw control.

mhkabir, 2016-05-27:
The Snapdragon's internal mag doesn't even have a released driver yet. 
Your testing is flawed, or the attitude estimator is not fusing magnetometer measurements.

mhkabir, 2016-05-27:
If you really want to check, see the raw mag values in the QGC Analyse widget (HIGHRES_IMU message)

julianoes, 2016-05-27:
@SamChenzx which commit did you test fly?

Note that the rotation option does not work on Snapdragon yet. I need to add it.

mhkabir, 2016-05-27:
@julianoes I suspect that it was the tone vs.  hmc5883l driver clash, which meant that the mag wasn't working? Fixed on master, right?

julianoes, 2016-05-27:
Correct, even though I wouldn't think something would take off without mag because the ekf2 doesn't initialize.

mhkabir, 2016-05-27:
So that's what was causing my fails before I switched to the Q estimator!

SamChenzx, 2016-05-28:
@mhkabir @julianoes  Today I use the latest pull and done more tests. 
Firstly, I removed the 3DR GPS/MAG module and cleared sensors calibration parameters, then tried to calibrate sensors, this time can't find GPS information and can't step into compass calibration, thus, make sure external compass on 3DR GPS module is used. But without GPS lock, sometimes heading will turn slowly. As I check the HIGHRES_IMU.x(y,x)mag, both direction, I do  360 degree rotation both directions of 3DR GPS module, the result shows as the follow:
[image]

[image]

[image]
Clearly, the scales has some bugs or my 3DR GPS module encounter a hardware issue.

Secondly,  as we attempt to takeoff outdoor with GPS locked, one or two motors always has lower PWM duty, but which motors, according to our tests, are decided by the angle between snapdragon flight and the direction of 3DR GPS module. The follow is a test result, only throttle is added, obviously, it can't really takeoff , or quadrotor will rotate fast and fall down.
[image]
With GPS locked, if I change GPS's direction, heading will slowly change according GPS to the correct direction and shows on QGC.

Thirdly, where I think the core problem is, as snapdragon flight is assembled on quadrotor with optical flow camera face down, the 3DR GPS module is assembled on with it's white heading mark the same direction with snapdragon flight's camera(X+ direction of MPU9250's accelerometer), their coordinate systems are inconsistent. Shown in the follow two pictures, 
[image]
[image]
But when calibration sensors, I leave all the Autopilot orientations "ROTATION_NONE", as in my  comprehension, they point to the same direction, if I'm wrong here, please inform me.
Besides, my snapdragon flight is a P2 board, the orientation of MPU9250 is different from P1 board in X and Y directions as shown in the follow:
[image]
So maybe some rotations of sensors should be set, but currently I have no idea how to set them correctly, please help!

mhkabir, 2016-05-28:
Ah. Likely your rotations are off. The mag and gyro are likely "fighting".
The compass is working now though (since you updated to master), as evident from the plots.

> Clearly, the scales has some bugs or my 3DR GPS module encounter a hardware issue.

How do you arrive at this conclusion?

@julianoes We really need to implement the rotation functions for the board and compass. This is the first P2 generation board I've seen, and it looks like the sensor orientation might be different.

julianoes, 2016-05-28:
> Besides, my snapdragon flight is a P2 board, the orientation of MPU9250 is different from P1 board in X and Y directions as shown in the follow:

Aha this is important feedback, thanks.

I suggest the following plan:
- In the short term: implement the rotations for MPU9250 and HMC5883, so that you can set the orientation (and adapt it to P1 vs. P2.)
- In the longer term: come up with some check that discovers if the board is P1 or P2 automagically and set the orientation accordingly.

SamChenzx, 2016-05-28:
I arrived at it since the compass is rotated 360 degrees every direction,
the output data will varies from 1to -1 theoretically, or after calibration
it should varies from 1 to -1.

SamChenzx, 2016-05-28:
@julianoes Julian, so the easiest way for me to fix this issue is to set the rotation same as P1 board and still keep the direction of 3DR GPS module's white front mark along with the direction of snapdragon flight's camera?

mhkabir, 2016-05-28:
@SamChenzx yep.

SamChenzx, 2016-05-28:
@mhkabir Thanks, I'll try rotation parameters on QGC.

SamChenzx, 2016-05-30:
@julianoes Julian, I've done four tests: 
1) Setting the accelerometer and gyro rotation parameter to ROTATION_YAW_270, but leave magnetometer ROTATION_NONE, let 3DR GPS's front mark along the direction of the snapdragon flight's camera.
2) Setting the accelerometer and gyro rotation parameter to ROTATION_YAW_270, but leave magnetometer ROTATION_NONE, let 3DR GPS's front mark along the Y- direction of the snapdragon flight's MPU9250(compare to P1 board).
3) Setting the accelerometer, gyro and magnetometer rotation parameter to ROTATION_YAW_270, let 3DR GPS's front mark along the direction of the snapdragon flight's camera.
4) Setting the accelerometer, gyro and magnetometer rotation parameter to ROTATION_YAW_270, let 3DR GPS's front mark along the Y- direction of the snapdragon flight's MPU9250(compare to P1 board).

Both result seems didn't work, as the output of PWM duty still has one or two channels less then others.
In each experiment, I clear the sensors parameter, and re-calibrate the sensors. Is that due to rotation function hasn't added yet or I set the wrong rotation parameter?

mhkabir, 2016-05-30:
@SamChenzx Not implemented in the drivers yet, sorry! I can do it today, if you're willing to test it.

SamChenzx, 2016-05-30:
@mhkabir I will test today, and want to sent snapdragon flight into the air before June. Thanks!

mhkabir, 2016-06-03:
@SamChenzx Could you please post a picture of your vehicle here?

SamChenzx, 2016-06-03:
@mhkabir 
I think maybe they are two different issues. Here is my vehicle now the GPS module and camera are towards the same direction.
[image]

mhkabir, 2016-06-03:
@SamChenzx OK, this looks correct. I'm testing with my own vehicle, give me a minute.

Can you confirm that the artificial horizon functions correctly in QGC with the MPU9250 running with ```-R 6```?

SamChenzx, 2016-06-03:
@mhkabir Yes, I set "-R 6" every time I flash the board. Even I have tried to set "-R 6" for the HMC5883.

mhkabir, 2016-06-03:
Just to be clear, you shouldn't need to do *anything* other than the ```-R 2``` or ```-R 6``` on the MPU9250 driver.
Please confirm whether the horizon behaves correctly in QGC with either of those rotation parameters.

SamChenzx, 2016-06-03:
@mhkabir Today I also set ROTATION_YAW90 in QGC when do the calibration, if just set "-R 6", heading will not toward the camera. Is that the problem? But I've noticed rotation in sensors can also work as Julian said.

mhkabir, 2016-06-03:
@SamChenzx Could you please just follow what I'm saying? No additional settings other than ```-R 6```.

SamChenzx, 2016-06-03:
@mhkabir If only -R 6 is set, I have takeoff once but the direction not correct. https://youtu.be/tkQ1cZjRAfA

mhkabir, 2016-06-03:
@SamChenzx OK, good. So it flies correctly atleast, as I had expected. Now can you remove the HMC5883 and fly again after recalibrating? Just using internal mag of MPU9250.

SamChenzx, 2016-06-03:
@mhkabir This video is shot two days ago, at that time the driver of internal mag hasn't released yet. I really want to know how you define the heading direction of a vehicle?

mhkabir, 2016-06-03:
Heading direction = Direction nose points towards

SamChenzx, 2016-06-03:
@mhkabir Yesterday is too late in the night for me to do more test. I tested just now with a new git clone of the Firmware, changed nothing, just make and upload to the snapdragon. 
Tried to set `df_mpu9250_wrapper start -R 6` or `df_mpu9250_wrapper start -R 2`, then carefully re-calibrate sensors before each test. Without set sensors ROTATION parameters in QGC, during calibration, the heading direction didn't point to the camera direction. Both time I rotate vehicle counterclockwise 90 degrees, the yaw drifts but the sensor data is correct. shown in the following plot:
I think the drift is the true issue now. 
The weird thing is if I set the `df_mpu9250_wrapper start -R 2` then arm motor and push up throttle, the quadrotor is still but the yaw is changed shown as the plot 2 last minute. This situation, I think one motor didn't spin due to it want to yaw itself shown in this [video](https://www.youtube.com/watch?v=m-bTLDBeHw8&feature=youtu.be): https://www.youtube.com/watch?v=m-bTLDBeHw8&feature=youtu.be

R6: counterclockwise 90 degrees
[image]

R2: counterclockwise 90 degrees last one minute push up throttle then push down
[image]

[px4.config_R2.txt](https://github.com/PX4/Firmware/files/298024/px4.config_R2.txt)
[px4.config_R6.txt](https://github.com/PX4/Firmware/files/298025/px4.config_R6.txt)

SamChenzx, 2016-06-03:
Forget to attach the mag calibration result:

[image]

SamChenzx, 2016-06-10:
@julianoes @mhkabir Help, can't stop drifting after takeoff! At my last post, the yaw drift seems a bug of 3DR GPS module, fixed that bug, I attempt to use Altitude control and Position control flight mode, both can takeoff stably, but then drift quickly, position control mode flight shows in this [video](https://youtu.be/XFzFmjOCK5g) https://youtu.be/XFzFmjOCK5g . Does It seems like GPS didn't lock position correctly, or is there something else didn't work? Please help.

julianoes, 2016-06-10:
> At my last post, the yaw drift seems a bug of 3DR GPS module, fixed that bug

What did you fix?

SamChenzx, 2016-06-10:
I just checked the cable, make sure it connect well, sometimes it still
drifts, maybe bugs in snapdragon. But my concern is that why it drifts so
quickly after takeoff?

julianoes, 2016-06-10:
@SamChenzx Can you share a log or screenshot that shows exactly what drifts and how? Thanks

mhkabir, 2016-06-10:
Yeah I've seen the drift issue as well. It's observable as a skewed artifical horizon in QGC. There is a random offset at boot which converges sometimes, and doesn't other times. I'm running the Q estimator. 

On airframes with high vibrations, you can see a similar effect after landing. The attitude estimate just badly diverges.

julianoes, 2016-06-10:
@mhkabir that is with an external mag?

SamChenzx, 2016-06-10:
Yes, I use the external mag, maybe I have the log file on snapdragon
flight, screen shot means the trajectory on QGC?

julianoes, 2016-06-10:
@SamChenzx either one might help, yes.

SamChenzx, 2016-06-10:
@julianoes Sorry for late. Here are the two logs, Altitude control mode and Position control mode respectively.
[px4log.zip](https://github.com/PX4/Firmware/files/309482/px4log.zip)

SamChenzx, 2016-06-11:
@julianoes I suddenly realized that, the log files I posted yesterday maybe not the exact log file I tested in the field. As I pull them from the "sess0*" directory, which means no GPS locked, but the useful log file I think should under the "2016-06-09" directory.

julianoes, 2016-06-13:
> As I pull them from the "sess0*" directory, which means no GPS locked

This is differently on Snapdragon. sess0* means no date/time set which only happens through wifi and NTP but not through GPS because `clock_settime()` is not supported on QURT.

julianoes, 2016-06-13:
@SamChenzx These are helpful logs. And on a first look it seems that the estimator goes wild, even just sitting on ground. You can see it by comparing GPS and GPOS velocities:

[image]

I checked the version that you flew and there have been fixes to ecl/ekf2 since, e.g.:
https://github.com/PX4/ecl/commit/f08aee0a7dab450252935c562c2d16125744336d
https://github.com/PX4/ecl/commit/21168f992996503b995094f2cf143dfe2d44469e

I would therefore suggest to flash latest master and test it by just having it sit on the ground and observing QGC or the logs and checking if the position estimate behaves.

SamChenzx, 2016-06-13:
> PX4/ecl@f08aee0
> PX4/ecl@21168f9

Thanks, I would use latest master and test today, will upload logs here.

SamChenzx, 2016-06-13:
@julianoes Just now I've done a test on the bench with latest master.
Without GPS signal, I can't have GPS related plot. But one issue always
confused me is the ATTITUDE_TARGET.body_yaw_rate. I use the virtual
joysticks on QGC for controller, when arm it, and push throttle up or down,
ATTITUDE_TARGET.body_yaw_rate will changed intensely while ATTITUDE.yaw
keep still. I think that's a reason for the PWM output goes weird, here are
the plot and log file.

​

​
 log001.px4log.zip
<https://drive.google.com/file/d/0B8KmoPJKKvZeVVBla1cwV3phUmc/view?usp=drive_web>
​

SamChenzx, 2016-06-13:
@julianoes As my knowledge, if there is no yaw changed, the ATTITUDE_TARGET.body_yaw_rate should gradually back to zero, but it didn't.

julianoes, 2016-06-13:
@SamChenzx: can you track your PWM troubles to another issue? This thread is already hijacked enough :)

SamChenzx, 2016-06-13:
@julianoes I think it comes from the same issue, as it can takeoff in position control mode, PWM output should be correct, but strange in manual flight mode.  The ATTITUDE_TARGET.body_yaw_rate confused me.

julianoes, 2016-07-27:
The initial issue is resolved since there are presumably only P2 boards around even though the silkscreen says P1.

See: http://dev.px4.io/advanced-snapdragon.html#do-i-have-a-p1-or-p2-board

closed 2016-07-27 by julianoes

SamChenzx, 2016-07-27:
Yes, and now my snapdragon works well, thanks for your help!