PX4/PX4-Autopilot #4654: Snapdragon: ekf2 doesn't init on field

julianoes, 2016-05-27:
The ekf2 didn't initialize successfully on the field today, even though GPS was supposedly good.
Need to investigate.

@priseborough FYI

priseborough, 2016-05-27:
Did you get a replay log for me?

julianoes, 2016-05-27:
It looks like one of the accuracy checks didn't succeed. I raised the threshold and got a home init from my balcony. I'll have to test on the field again.

julianoes, 2016-05-27:
From looking at the log the problem is the field sacc (speed accuracy) which never really goes below 1.

Presumably this value is flawed (or different) because I'm using an older uBlox which sends NAV-SOL and not NAV_PVT.

priseborough, 2016-05-27:
It's normally caused by poor signal to noise ratio and is nearly always caused by either interference from nearby electronics, a  poor antenna or signal blockage from nearby structures and foliage. If you run the same receiver on a pixhawk, what does it produce? Anyway you can increase the threshold to 1.5 m/s via the EKF2_REQ_SACC parameter, but if the GPS speed is that noisy, the loiter stability will be degraded.

julianoes, 2016-05-27:
Good call, I'll cross test this GPS with a Pixhawk. The field was open, so that shouldn't be the issue.

julianoes, 2016-05-27:
I cross tested this on Pixhawk using two different 3DR GPSs and got very similar results for sAcc. This was on the Balcony, I'll have to retest on an open field as well.

Has the check for sacc been added recently, or has it been there for a while?

julianoes, 2016-05-27:
More sAcc data:

Snapdragon with wrapped wifi antenna: 0.7 .. 1.1 m/s
Snapdragon with wrapped camera (front part): 0.5 .. 1.0 m/s
Snapdragon wrapped (wifi free): 0.4 .. 1.0 m/s
Snapdragon wrapped and wifi wrapped: 0.3 .. 0.8 m/s

LorenzMeier, 2016-05-28:
So essentially its a GPS jammer like many of the computing boards are.

julianoes, 2016-05-28:
Yes, I'm now switching to the M8N (again) and moving it further up while moving the FrSky antennas that seemed to previously interfere with M8N further away. Lets see how this goes.

dagar, 2016-05-28:
I'm not sure what your setup looks like and which parts have been wrapped, but I had a lot of interference from USB3 on a Tegra TX1.

mhkabir, 2016-05-29:
@dagar I just ran into the same issue yesterday on a TX1. Did you manage to fix it? Previous boards I've worked with are far worse though... Often disabling the Ethernet helps, because of the clock speed.

Also running into this on the Snapdragon (lucky me!)
@julianoes Your test suggests that shielding the board actually works though. Could you describe how?

julianoes, 2016-05-30:
@mhkabir I just put some layers of tinfoil between Snappy and GPS (and some electrical isolation with tape to prevent shorts), that's a very poor man's solution. Better would be a 3D printed case with a layer of copper paint/spray on the inside.

dagar, 2016-05-30:
I also had to cover the USB 3 camera and cable in foil before my m8 was usable.

mhkabir, 2016-05-30:
@dagar @julianoes Thanks for the tips! I've made small copper shielded cases on my big copter to hold all the components like the TX1, so hopefully that fixes it :)

julianoes, 2016-07-13:
I ended up raising the GPS. If the GPS is just a few centimeters higher, the GPS issues are mostly resolved.

closed 2016-07-13 by julianoes