PX4/PX4-Autopilot #4658: Servo's unresponsive after transition

sanderux, 2016-05-27:
We are tracking down a problem with unresponsive servo's after transition to fixed wing on vtol's. There are currently 2 identical reports that share the following behavior;

- Servo's work fine on the ground, even under load.
- After transition they behave eratic (switch to a position and then hold there, or not respond at all)
- RX starts behaving strange in some occasions

Logs:
Tailsitter: http://logs.uaventure.com/view/7ZediDktxQzcBsZ2wBizgm
QuadPlane: http://logs.uaventure.com/view/5MFXSfPmYwrUFxuCDnQj7T

The platforms are similar in the following;
- Digital servo's
- Delta configurations
- PWM rail powered by bec from esc (pusher / puller)

The platforms differ in the following;
- Tailsitter vs quadplane
- Pixracer vs pixhawk
- OrangeRX vs FrSky receivers
- 4s vs 8s
- master vs custom attitude controller (not using mixer, producing PWM output directly)

It remains unsure if these problems are hardware or software related

@CarlOlsson @tumbili

sanderux, 2016-05-27:
The deltaquad v1 flew fine, v2 exhibited these problems. Key differences in the 2 airframes;
- mauch power module vs 3dr
- digital vs analog servo's

V2 flew on the same software as the V1 and exhibited the same problems.

LorenzMeier, 2016-05-27:
Did the system which had issues have digital servos?

sanderux, 2016-05-27:
Yes

LorenzMeier, 2016-05-27:
Which log shows the issues? I suspect your servos are too smart and either go into timeout or programming mode.

sanderux, 2016-05-27:
both logs show the same issues, the first log is for a tailsitter from @CarlOlsson and the second is my v2 deltaquad

sanderux, 2016-05-28:
Today i successfully flew the deltaquad V1 on the same software that had the servo issues.
http://logs.uaventure.com/view/iVKdZ4ijG8jMmgFij9Q6A9

Now i will incrementally update hardware to identify the problem.
I did notice the servo's reacted to the proximity of the telemetry transmitter on the servo wires.
@CarlOlsson I would suggest you test fly without temelletry module

CarlOlsson, 2016-05-29:
Glad to hear it. We will do the same. Tomorrow we will replace all electronics with a complete setup from an old airplane and then incrementally replace hardware. Yes you are right, with a high power telemetry module servos may gitter if they are to close, we are running low output power though.

CarlOlsson, 2016-05-29:
@sanderux
Did you still have problems with RC signal?

sanderux, 2016-05-29:
No, but i did fly the old setup, that uses a different receiver

sanderux, 2016-06-14:
CG issue

closed 2016-06-14 by sanderux