PX4/PX4-Autopilot #4660: Standard VTOL reports MAV_TYPE as MAV_TYPE_QUADROTOR

DonLakeFlyer, 2016-05-27:
Instead of MAV_TYPE_VTOL_QUADROTOR. Which in turn causes QGC to show the wrong things. I didn't check all the other VTOL types for correctness. But those should be checked as well.

LorenzMeier, 2016-05-27:
Which airframe did you select?

DonLakeFlyer, 2016-05-27:
Standard VTOL

LorenzMeier, 2016-05-27:
It reports two different VTOL types - what is going on with QGC?

Console is correct on PX4:

[image]

Param value is incorrect in QGC:

[image]

Value via MAVLink is correct:

[image]

dagar, 2016-05-27:
This is probably why setting the airframe to a fixed wing for the first time results in the safety screen still showing multicopter icons and settings after the auto reboot and reconnect.

LorenzMeier, 2016-05-27:
I filed an issue for QGC and will close this one.

LorenzMeier, 2016-05-27:
https://github.com/mavlink/qgroundcontrol/issues/3422

DonLakeFlyer, 2016-05-27:
QGC pays no attention to the MAV_TYPE parameter. It gets MAV_TYPE from the heartbeat and sets it once from there.

DonLakeFlyer, 2016-05-27:
> This is probably why setting the airframe to a fixed wing for the first time results in the safety screen still showing multicopter icons and settings after the auto reboot and reconnect.

Nope (MAV_TYPE from hearbeat). Does that repro on latest builds. If so can yo uput and issue in?

DonLakeFlyer, 2016-05-27:
When you change airframe from QGC and then the board reboots the first heartbeat QGC gets will have a MAV_TYPE of the old type and then the second heartbeat it gets will have the correct new MAV_TYPE. QGC constructs the Vehicle from the first heartbeat with the incorrect MAV_TYPE and then all hell breaks loose from then on.

dagar, 2016-05-27:
We can either move mavlink to the end of the init script or use the boot_complete command the mavlink app already has.

LorenzMeier, 2016-05-27:
@dagar @DonLakeFlyer This should fix it. I didn't get to test it yet: https://github.com/PX4/Firmware/pull/4666

closed 2016-05-28 by DonLakeFlyer