PX4/PX4-Autopilot #4671: Plane model for Gazebo

LorenzMeier, 2016-05-28:
@hsu This is to track the bug fixing of the plane model in Gazebo. @tfoote @tumbili @CarlOlsson FYI.

hsu, 2016-05-31:
I need help understanding the collision shapes used for the [plane model](https://github.com/PX4/sitl_gazebo/blob/master/models/plane/plane.sdf)

[image]

why they are very different from the visual.  Thanks.

hsu, 2016-05-31:
Change to collision models to stabilize contacts

https://github.com/PX4/sitl_gazebo/pull/23

LorenzMeier, 2016-06-01:
The collision model is just inexperience / hacking on our part. I guess we should look at doing a clean new model.

hsu, 2016-06-02:
got it. If you like, I can take a rough pass at the collision shape. I take it's not super critical how the plane makes contact with surrounding environments, except mabye 1+ box-shaped collision shape that envelopes the whole visual model with stable contacts with the ground? Do you plan to use the wheels for runway / taxiway operations?

LorenzMeier, 2016-12-28:
Finished / working.

closed 2016-12-28 by LorenzMeier