PX4/PX4-Autopilot #4674: We need a mode witch can desend without GPS.

xn365, 2016-05-28:
I lost my quad due to RC LOSS.
It had no GPS, So when RC LOST,it just fly away.
Logs:
http://logs.uaventure.com/view/jLdEbnkBeEjevq8geGb5hj

kd0aij, 2016-05-28:
Very sorry to hear that. How do you know the cause was RC loss? Was the
quad really "lost" or did you get the log from the SD card afterward?

xn365, 2016-05-28:
It finally landed by a tree, so i got the logs.
All RC_RAW_VALUES are null in logs.

kd0aij, 2016-05-28:
I was afraid that might happen; never felt like putting it to the test myself.
How did you arm the quad? The log appears to show that you never had an RC link.

xn365, 2016-05-29:
This is test log:
[10:27:46.487 - COMP:1] Info: SUCCESS: Executed CMD: 245
[10:27:52.399 - COMP:1] Info: ARMED by RC
[10:27:52.732 - COMP:1] Info: [blackbox] /fs/microsd/log/sess018
[10:27:53.095 - COMP:1] Info: [blackbox] recording: log001.px4log
[10:27:54.268 - COMP:1] Info: TAKEOFF DETECTED   
[10:28:07.622 - COMP:1] Critical: failsafe mode on      -> When I pull out the receiver,QGC display this line.Mode chang to AUTO|LAND.

Then,I am waiting for landing,but nothing happened.

xn365, 2016-06-01:
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L344
I think navigator will not working When we don't have gps.
Could AUTOLAND or LANDGPSFAIL mode run in this case?
https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L615
We will chang to DESCEND mode in the case of RC lost and have no GPS.
But this line will never be reached when we have no GPS :https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L566

julianoes, 2016-06-01:
@xn365 Thanks for reporting!

You found exactly what I fixed in #4559, it's time to test this and get it in!

julianoes, 2016-06-21:
Done in #4559.

closed 2016-06-21 by julianoes