PX4/PX4-Autopilot #4680: Out of Memory during Cal Calibration

tops4u, 2016-05-28:
I was attaching an external Compass and did several Mag Calibrations. Three Side Calibration did not finish (all Icons Green, text ok, but Progress bar did not reach end and OK Button was not shown). I then changed to 6 Side Calibration and tried again. This time I got the following Error:

```
Start the individual calibration steps by clicking one of the buttons above.
[cal] calibration started: 2 mag
[cal] Skipped scale calibration, sensor 1
[cal] ERROR: out of memory
[cal] calibration failed: mag
```

QGC 2.9.7b - 60 / Fimware shown in QGC is 1.6.4dev (Master from today). 

Despite this disconnecting a Mag and redoing Calibration did not reset the MagID. Without manualle erasing all Mag Infos it was not possible to identify the newly externally attached mag. Missing Mag was also not shown in the Sensors in System Summary Screen, Mags shown with green Indication despite that I had disconnected the external Mag.

LorenzMeier, 2016-05-29:
This PR should help with that:
https://github.com/PX4/Firmware/pull/4687

LorenzMeier, 2016-05-30:
Fixes are on master. Can you retry?

closed 2016-05-30 by LorenzMeier

tops4u, 2016-05-31:
I'll try. Not sure I can redo it....

silkstone, 2016-08-24:
Hi @LorenzMeier, I am having the exact same issue on QGC. Seems to have occurred when I upgraded Pixracer firmware to 1.4.1 as it worked ok before this. Am trying to replicate the Caipriniha VTOL. Any suggestions as to why this is happening? 

[cal] calibration started: 2 mag
[cal] Skipped scale calibration, sensor 1
[cal] ERROR: out of memory
[cal] calibration failed: mag

LorenzMeier, 2016-08-24:
@silkstone Are you sure your board is getting enough power? Otherwise please re-flash v1.4.2 via QGC. Release notes are here: https://github.com/PX4/Firmware/releases/tag/v1.4.2